ArduPilot/ardupilot #25257: Problem when connect AirSim with Ardupilot, "no sensor message ..."

iagupov, 2023-10-13:
## Bug report

When I run AirSim binaries on Windows 10 and then ArduPilot on WSL2 Ubuntu 22.04 to connect them, I get this error
"no sensor message received in last 1s, error - bad file descriptor, resending servos"
and I can't link my AirSim with ArduPilot. I followd all the procuderes I could found, but it didn't help.

[image]

Also in settings.json (for AirSim binaries) I can't use “127.0.0.1” both for LocalHostIp and UdpIp as [guide](https://ardupilot.org/dev/docs/sitl-with-airsim.html#launch-copter-sitl) says. Otherwise UE4 binareas freezes and don’t move when run. So for UdpIp I use the one obtained from “ip addr show eth0”.

If anybody encountered this issue please help.

**Version**
Windows 10
WSL 2 with Ubuntu 22.04
Ardupilot 4.4.1
AirSim 1.8.1

**Platform**
[x] Copter

eziosoft, 2023-11-02:
I have exactly the same problem

devnim, 2023-12-27:
I get the same error message "No sensor message received in last 1s, error - bad file descriptor, resending servos"