ArduPilot/ardupilot #25265: BLHeli S and DShot in 4.5.0dev

keithluneau, 2023-10-15:
## Bug report

**Issue details**

I'm flying an Atomrc Swordfish PNP, using their F405-Navi flight controller and their BLHeli ESCs. I'm using DShot600 protocol, with the stock BLHeli S firmware on the ESCs. This has worked well in previous versions of firmware, verified up to 4.4.2 beta-1. After updating to 4.5.0dev, the ESCs are not happy. Once the FC finishes booting, both motors start to beep and stutter when they should be sitting idle. I tried to DShot150 and 300 protocols with no change. Falling back to 4.4.2 beta-1 or earlier resolves the issue.

For what it's worth, a different plane using a Matek H743-WLite flight controller with a single motor and 32bit BLHeli ESC works fine with 4.5.0dev, with DShot or BDShot. 

**Version**
4.5.0dev downloaded today, 10/14/23

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Airplane, Atomrc Swordfish

**Hardware type**
Atomrc F405-Navi

**Logs**
None to provide, but happy to get whatever I can to help troubleshoot/pinpoint the issue on my end if needed.

Hwurzburg, 2023-10-15:
can ou specify the escs mentioned?

keithluneau, 2023-10-15:
> can you specify the escs mentioned?

Yes, they are "ATOMRC BLS 30A 5VBEC V2 ESC"
https://atomrc.com/products/atomrc-bls-30a-5vbec-v2-esc

Hwurzburg, 2023-10-16:
@andyp1per Keith did these tests:
[image]

andyp1per, 2023-10-16:
@keithluneau can you send me the output of ```@SYS/timers.txt``` for both 4.4 and 4.5? You can get this via mavproxy or possibly mission planner

keithluneau, 2023-10-16:
> @keithluneau can you send me the output of `@SYS/timers.txt` for both 4.4 and 4.5? You can get this via mavproxy or possibly mission planner

Sure thing! I do note a difference in frequency with both set up for DShot600.

4.4.2beta1:
TIMERV1
TIM2  CLK=  84Mhz MODE=  PWM FREQ=       8 TGT=       8
TIM3  CLK=  84Mhz MODE=DS600 FREQ= 4941176 TGT= 4800000
TIM4  CLK=  84Mhz MODE=  PWM FREQ=       8 TGT=       8
TIM1  CLK= 168Mhz MODE=  PWM FREQ=       8 TGT=       8

4.5.0dev:
TIMERV1
TIM2  CLK=  84Mhz MODE=  PWM FREQ=       8 TGT=       8
TIM3  CLK=  84Mhz MODE=DS600 FREQ= 5250000 TGT= 4800000
TIM4  CLK=  84Mhz MODE=  PWM FREQ=       8 TGT=       8
TIM1  CLK= 168Mhz MODE=  PWM FREQ=       8 TGT=       8

andyp1per, 2023-10-16:
the second is clearly wrong. what is SERVO_DSHOT_ESC set to in both?

keithluneau, 2023-10-16:
It is set to 0 on both.

keithluneau, 2023-10-16:
...and 4.4.2 with SERVO_DSHOT_ESC = 2

```
TIMERV1
TIM2  CLK=  84Mhz MODE=  PWM FREQ=       8 TGT=       8
TIM3  CLK=  84Mhz MODE=DS600 FREQ= 7000000 TGT= 6600000
TIM4  CLK=  84Mhz MODE=  PWM FREQ=       8 TGT=       8
TIM1  CLK= 168Mhz MODE=  PWM FREQ=       8 TGT=       8
```

andyp1per, 2023-10-16:
this is actually only a reporting problem. I'll post a PR to fix this, but the timing information is the same.

andyp1per, 2023-10-16:
This fixes the reporting - https://github.com/ArduPilot/ardupilot/pull/25280