ArduPilot/ardupilot #25266: Using heading decomposition attitude angle about loiter control

lixiaowei123, 2023-10-15:
[image]
[image]
this should probably be based on the desired heading not the current heading,but I did use the actual heading recorded in the log, so there is a problem with position control

lthall, 2023-10-15:
I put that note in there to remind me to come back and examine this decision.

Using desired heading results in a lower noise target attitude command. However, it also means that loss of yaw control will mean loss of position control.

My initial thought is that we are probably better off using current heading and accepting the noisier attitude target so that we maintain position control when we have lost yaw control.

So the fix in this case would be to remove the comment.

lixiaowei123, 2023-10-15:
> I put that note in there to remind me to come back and examine this decision.
>
> Using desired heading results in a lower noise target attitude command. However, it also means that loss of yaw control [...]

When the actual heading is lost, the actual heading will be 0. Then, directly determine whether it is possible to modify the target heading to 0 when the heading is lost

lthall, 2023-10-15:
>> When the actual heading is lost, the actual heading will be 0. Then, directly determine whether it is possible to modify the target heading to 0 when the heading is lost

Sorry but I don't understand what you are saying. If we loose our actual heading we can't navigate at all. If we loose heading control we can navigate.

I don't see any situation where we can benefit from setting the heading to zero.

lixiaowei123, 2023-10-15:
@lthall Why not use the target heading，

lixiaowei123, 2023-10-15:
px4 use target heading

lixiaowei123, 2023-10-15:
PX4 target heading

lthall, 2023-10-15:
I explained that:

> Using desired heading (target heading) results in a lower noise target attitude command. However, it also means that loss of yaw control will mean loss of position control. My initial thought is that we are probably better off using current heading and accepting the noisier attitude target so that we maintain position control when we have lost yaw control.

> px4 use target heading

I aspire to much higher performance than PX4 is capable of. This is just one more list thing on the list of things they do wrong.

I am closing this as I can see this is not going anywhere.

closed 2023-10-15 by lthall

lixiaowei123, 2023-10-15:
Thank you ,Looking forward to you making better loiter brakes, similar to DJI

lthall, 2023-10-15:
I suspect your loiter breaking issues is a result of poor tuning and setup and nothing to do with the code. However, this isn't a support forum. You would get better results and waist less time by posting questions on the support forums.