ArduPilot/ardupilot #25275: loiter mode (calculate desire speed bug)

lixiaowei123, 2023-10-16:
[image]
BUG: !is_zero(desired_speed)  is const true

Should be modified to：
[image]

xianglunkai, 2023-10-16:
I don't seem to be making any mistakes, because the natural directed speed>=0.
 Additionally, apm has an impact on is_ Zero has been overloaded, which will result in a true zero value situation！

lthall, 2023-10-16:
The desired_speed can't be negative so there should be no change between 
`!is_zero(desired_speed) `
and 
`is_positive(desired_speed)`

I don't know anything about overloading is_Zero. This is something that @peterbarker could answer very quickly.

Peter, is this something that we should change?

lixiaowei123, 2023-10-16:
desired_speed>=0.0f，but  You can print and see if it has a very small value，for example  desired_speed=0.00001, result in !is_zero(desired_speed) ==1; Always running，

lthall, 2023-10-16:
0.00001 is not zero. Why do you expect this to not run? It should only not run if it is zero.

lixiaowei123, 2023-10-16:
desired_speed==0.0f；is_zero(desired_speed)==1；buf desired_speed>0.0f   is_zero(desired_speed)==1 result in !is_zero(desired_speed) ==1

lixiaowei123, 2023-10-16:
desired_speed==0.0f；is_zero(desired_speed)==1；buf desired_speed>0.0f, is_zero(desired_speed)==0 result in !is_zero(desired_speed) ==1

lixiaowei123, 2023-10-16:
I didn't move the joystick, and at this point, the expected speed should be 0; 
_desired_accel.x=0 .0f
_desired_accel.y=0.0f; 
howver 
_predicted_accel.x!=0.0f
_predicted_accel.y!=0.0f

lixiaowei123, 2023-10-16:
I have thought for a moment and should modify it to the following situation to solve the essential problem：
if((abs(_desired_accel.x)==0.0f)&&((abs(_desired_accel.y)==0.0f)))
{
  _predicted_accel.x==0.0f
  _predicted_accel.y==0.0f
}

lixiaowei123, 2023-10-16:
[image]
[image]

lixiaowei123, 2023-10-16:
Also, when the expected speed is equal to 0, it should be cleared to zero, as indicated below：
void AC_Loiter::calc_desired_velocity(bool avoidance_on)
{
    float ekfGndSpdLimit, ahrsControlScaleXY;
    AP::ahrs().getControlLimits(ekfGndSpdLimit, ahrsControlScaleXY);

    const float dt = _pos_control.get_dt();

    // calculate a loiter speed limit which is the minimum of the value set by the LOITER_SPEED
    // parameter and the value set by the EKF to observe optical flow limits
    float gnd_speed_limit_cms = MIN(_speed_cms, ekfGndSpdLimit * 100.0f);
    gnd_speed_limit_cms = MAX(gnd_speed_limit_cms, LOITER_SPEED_MIN);

    float pilot_acceleration_max = angle_to_accel(get_angle_max_cd() * 0.01) * 100;

    // range check dt
    if (is_negative(dt)) {
        return;
    }
    // get loiters desired velocity from the position controller where it is being stored.
    const Vector3f &desired_vel_3d = _pos_control.get_vel_desired_cms();
    Vector2f desired_vel{desired_vel_3d.x,desired_vel_3d.y};

    //TODO：
    if(_desired_accel.is_zero())
    {
      _predicted_accel.x=0.0f;
      _predicted_accel.y=0.0f;
    }

    // update the desired velocity using our predicted acceleration
    desired_vel.x += _predicted_accel.x * dt;
    desired_vel.y += _predicted_accel.y * dt;

    Vector2f loiter_accel_brake;
    float desired_speed = desired_vel.length();

//    if (!is_zero(desired_speed))
    if (desired_speed>0.0f)
    {
        Vector2f desired_vel_norm = desired_vel / desired_speed;

        // calculate a drag acceleration based on the desired speed.
        float drag_decel = pilot_acceleration_max * desired_speed / gnd_speed_limit_cms;

        // calculate a braking acceleration if sticks are at zero
        float loiter_brake_accel = 0.0f;
        if (_desired_accel.is_zero()) {
            if ((AP_HAL::millis() - _brake_timer) > _brake_delay * 1000.0f) {
                float brake_gain = _pos_control.get_vel_xy_pid().kP() * 0.5f;
                loiter_brake_accel = constrain_float(sqrt_controller(desired_speed, brake_gain, _brake_jerk_max_cmsss, dt), 0.0f, _brake_accel_cmss);
            }
        } else {
            loiter_brake_accel = 0.0f;
            _brake_timer = AP_HAL::millis();
        }
        _brake_accel += constrain_float(loiter_brake_accel - _brake_accel, -_brake_jerk_max_cmsss * dt, _brake_jerk_max_cmsss * dt);
        loiter_accel_brake = desired_vel_norm * _brake_accel;

        // update the desired velocity using the drag and braking accelerations
        desired_speed = MAX(desired_speed - (drag_decel + _brake_accel) * dt, 0.0f);
        desired_vel = desired_vel_norm * desired_speed;
    }
    else
    {
    	//clear zero,cannot affect radio control joystick(roll||pitch)
    	if (_desired_accel.is_zero())
    	{
    		_desired_accel.x=0;
    		_desired_accel.y=0;
            desired_vel.x=0;
            desired_vel.y=0;
        	_brake_accel=0.0f;
           	loiter_accel_brake.x=0;
            loiter_accel_brake.y=0;
    	}

    }
    // add braking to the desired acceleration
    _desired_accel -= loiter_accel_brake;

    // Apply EKF limit to desired velocity -  this limit is calculated by the EKF and adjusted as required to ensure certain sensor limits are respected (eg optical flow sensing)
    float horizSpdDem = desired_vel.length();
    if (horizSpdDem > gnd_speed_limit_cms) {
        desired_vel.x = desired_vel.x * gnd_speed_limit_cms / horizSpdDem;
        desired_vel.y = desired_vel.y * gnd_speed_limit_cms / horizSpdDem;
    }

#if !APM_BUILD_TYPE(APM_BUILD_ArduPlane)
    if (avoidance_on) {
        // Limit the velocity to prevent fence violations
        // TODO: We need to also limit the _desired_accel
        AC_Avoid *_avoid = AP::ac_avoid();
        if (_avoid != nullptr) {
            Vector3f avoidance_vel_3d{desired_vel.x, desired_vel.y, 0.0f};
            _avoid->adjust_velocity(avoidance_vel_3d, _pos_control.get_pos_xy_p().kP(), _accel_cmss, _pos_control.get_pos_z_p().kP(), _pos_control.get_max_accel_z_cmss(), dt);
            desired_vel = Vector2f{avoidance_vel_3d.x, avoidance_vel_3d.y};
        }
    }
#endif // !APM_BUILD_ArduPlane

    // get loiters desired velocity from the position controller where it is being stored.
    Vector2p target_pos = _pos_control.get_pos_target_cm().xy();

    // update the target position using our predicted velocity
    target_pos += (desired_vel * dt).topostype();

    // send adjusted feed forward acceleration and velocity back to the Position Controller
    _pos_control.set_pos_vel_accel_xy(target_pos, desired_vel, _desired_accel);
}

lixiaowei123, 2023-10-16:
[code.pdf](https://github.com/ArduPilot/ardupilot/files/12912641/code.pdf)

lixiaowei123, 2023-10-16:
@xianglunkai @lthall
 I have made many modifications here, but I feel that some very detailed things have not been properly handled.
Could you please take a closer look at whether it's reasonable? There were indeed issues with the previous one

lthall, 2023-10-16:
> desired_speed==0.0f；is_zero(desired_speed)==1. !is_zero(desired_speed) ==0

no:

lthall, 2023-10-16:
Ok, I think I understand why you are confused.

I think you are expecting the desired speed to be zero any time you have the sticks in the middle. This is not correct.

Loiter is a drag based kinematic model. So the aircraft slows down proportional to it's speed (linear not squared relationship). Lets say the speed halves every second, it never gets to zero but it keeps getting smaller. So when you zero the sticks you are not commanding zero velocity, you are commanding zero acceleration and letting drag change the speed.

The reason we use is_zero, is_negative, and is_positive is because they check that the number is zero to within the a couple times the float's minimum size. So is_zero(0.000000000000000000000000000000001) or some small number will be true.

> Also, when the expected speed is equal to 0, it should be cleared to zero, as indicated below：

No, this is not true and is one of the basic principles of the command model. I suspect there are a few aspects of how loiter works that you have not fully understood. The _predicted_accel is not equal to the _desired_accel to account for the slower response time of the attitude controller. The desired velocity is not zero when you zero the sticks because you are commanding zero acceleration and waiting for the drag model to slow the aircraft.

lixiaowei123, 2023-10-16:
I understand this. If we first operate the pitch with full lever and then return to the center,the expected acceleration of the joystick at this time is zero，The drone begins to brake and decelerate。At that time, the brakes were completed，The expected speed in the end must be zero，
howver,     
 _predicted_accel.x!=0.0f;
  _predicted_accel.y!=0.0f;
Is this reasonable？

lthall, 2023-10-16:
Yes but the _predicted_accel will be very small.

lixiaowei123, 2023-10-16:
Can the predicted acceleration be reset to zero? If the expected acceleration is zero, the attitude error will be relatively large, which will cause significant speed

desired_vel.x += _predicted_accel.x * nav_dt;
desired_vel.y += _predicted_accel.y * nav_dt;
desired_vel will be very max

lthall, 2023-10-16:
> Can the predicted acceleration be reset to zero?

Yes but that would not be correct. 

> desired_vel.x += _predicted_accel.x * nav_dt;
desired_vel.y += _predicted_accel.y * nav_dt;
desired_vel will be very max

You can see that your interpretation of this problem is not correct. If you were correct the aircraft would not stop when we let go of the sticks and would continue to go faster and faster. It does not do what you are saying it will do.

lixiaowei123, 2023-10-16:
> Yes but that would not be correct.
>
>
> You can see that your interpretation of this problem is not correct. If you were correct the aircraft would not stop when we let go of the sticks and would [...]

> Yes but that would not be correct.
>
>
> You can see that your interpretation of this problem is not correct. If you were correct the aircraft would not stop when we let go of the sticks and would [...]

No, when the expected acceleration of the joystick is zero, the expected speed should not be from the predicted acceleration. At this point, the expected speed decreases with the braking process, and the aircraft will eventually come to a stop

lthall, 2023-10-16:
Well given I wrote loiter and you didn't I don't think you are in a position to tell me what it should or should not do.

Please do not open any more issues. You are waisting developer time.

closed 2023-10-16 by lthall