ArduPilot/ardupilot #25290: EKF3 bad wind estimate

tridge, 2023-10-17:
We have 2 logs where EKF3 gets silly wind estimates immediately after forward transition on a quadplane without airspeed sensor:
http://uav.tridgell.net/tmp/EKF3-wind/log_23_2023-10-2-18-28-58.bin
[image]
http://uav.tridgell.net/tmp/EKF3-wind/2023-10-08%2011-57-08_crassh.bin
[image]

both cases led to crashes. no replay log yet

timtuxworth, 2023-10-17:
Mine is a new plane (XuWing VT-BIrd Tilt Rotor) that has not been fully set up yet. It has an airspeed sensor but ARSPD_USE = 0 because I haven't be able to fly a proper calibration on it yet - similarly with the magfit, my next step after transition was to do a magfit flight, but I hadn't been able to do that yet.

Rolf-G, 2023-10-18:
@tridge: Sorry, perhaps I hadn't made it clear enough on Discord that I could reproduce the oscillation with another flight and LOG_REPLAY = 1 and LOG_DISARMED = 1. 
Here is the link:  https://magentacloud.de/s/RXMG7ykFD4rgFeR

LzREngineering, 2025-04-19:
Could this be a related problem to what I was seeing in this thread: [https://discuss.ardupilot.org/t/airspeed-ctun-as-is-opposite-to-expected/125091/13](https://discuss.ardupilot.org/t/airspeed-ctun-as-is-opposite-to-expected/125091/13)

In that instance, when the airspeed sensor is disabled (for whatever reason), the estimated windspeed is being subtracted, rather than added to the estimated airspeed, causing some very strange readings. I have still not found a solution (other than hoping my real airspeed sensor never fails!)