ArduPilot/ardupilot #25293: EKF3 calculate _accel_ef_ekf bug(accel level cal)

lixiaowei123, 2023-10-18:
BUG Version:
              ardupilot 4.3.3 Previous version
Now this version  4.3.3  has been fixed，cause :Acceleration level calibration，My submission was blocked a few days ago，I really can't figure out what good results would be if you didn't ask about right or wrong like this。I hope the two responsible people publicly apologize to me
[https://github.com/ArduPilot/ardupilot/issues/25279](url)

This  is 4.3.2 version

[image]

This  is 4.3.3 version
[image]
This serious bug can cause abnormal acceleration level calibration, abnormal landing and locking of the aircraft

Even more terrifying was the update of the records, but it was not explained

[image]

lthall, 2023-10-18:
This is a great example why we close issues that do not clearly articulate the problem.

We have many dedicated and talented users who are willing to clearly describe the problem and provide data from their own tests and clear instructions on how to replicate the problem. If you had made a basic reasonable effort before making your issue you may have realised it was already fixed. If you were clear about the problem then we may have been able to tell you it was already fixed.

I am closing this issue because like all the other issues you have opened, it does not meet the standards required. 

Please read and follow the instructions you deleted:

**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

closed 2023-10-18 by lthall

lixiaowei123, 2023-10-18:
> This is a great example why we close issues that do not clearly articulate the problem.
> 
> We have many dedicated and talented users who are willing to clearly describe the problem and provide data from their own tests and clear instructions on how to replicate the problem. One of those people will create a issue that lets us focus on fixing the problem.
> 
> If you had made a reasonable effort before making your issue you may have realised it was already fixed. If you were clear about the problem then we may have been able to tell you it was already fixed. In any case it is clear that you didn't check to see if your problem was still present in master and that is a complete lack of due diligence.
> 
> I am closing this issue because like all the other issues you have opened, it does not meet the basic standards required.
> 
> Please read and follow the instructions you deleted:

Thank you for your patient response, but I feel that you have resolved this bug and at the very least you need to explain it

lthall, 2023-10-18:
Please post support questions to the discuss forums (http://discuss.ardupilot.org/).