ArduPilot/ardupilot #25316: EKF_Buffer: there is a bug in ekf_imu_buffer::push_youngest_element(const void *element)

luweiagi, 2023-10-19:
Hi, in ekf_imu_buffer::push_youngest_element(const void *element):
https://github.com/ArduPilot/ardupilot/blob/5b34a9bf1370186c1aafe7f9f35dc77235401a9f/libraries/AP_NavEKF/EKF_Buffer.cpp#L159
```
void ekf_imu_buffer::push_youngest_element(const void *element)
{
    if (!buffer) {
        INTERNAL_ERROR(AP_InternalError::error_t::flow_of_control);
        return;
    }
    // push youngest to the buffer
    _youngest = (_youngest+1) % _size;
    memcpy(get_offset(_youngest), element, elsize);
    // set oldest data index
    _oldest = (_youngest+1) % _size;
    if (_oldest == 0) {
        _filled = true;
    }
}
```
, I think 
```
    if (_oldest == 0) {
        _filled = true;
    }
```
should be modified as:
```
    if (_oldest == 1 || _size == 1) {
        _filled = true;
    }
```

Why?

because when push, the first element is pushed into pos 1 NOT pos 2:
```
_youngest = (_youngest+1) % _size;  // here _youngest is initialized as 0, so the first elem pos is 1 not 0
```

so when _oldest is 0, there is N-1 elem not N elem in buffer:
` _oldest = (_youngest+1) % _size;` 
in fact, the buffer is not full.

amilcarlucas, 2023-10-19:
Even better than a bug report is a pull request. Can you make one?

luweiagi, 2023-10-20:
> Even better than a bug report is a pull request. Can you make one?

OK, I have make a PR:
fix imu_buffer _filled=true logic in EKF_Buffer.cpp #25323
https://github.com/ArduPilot/ardupilot/pull/25323

peterbarker, 2024-04-10:
Thanks @luweiagi , real bug and I merged a slightly different fix + regression tests as part of the other PR.

closed 2024-04-10 by peterbarker

luweiagi, 2024-04-18:
@peterbarker @rmackay9 Hi, I think the following fix is more correct:

PR #26696 (2024-04-10): Correct off-by-one problem in EKF IMU ringbuffer
This is an alternate solution to a bug @luweiagi found in the EKF buffer code.

His PR is here: https://github.com/ArduPilot/ardupilot/issues/25316

My formulation here switches the determination of whether the buffer has ever been full to looking for us wrapping the new offset, rather than looking at the old offset.

I think all of these approaches are bogus; if we advance the oldest pointer then we may never actually be "full".  I think the current behaviour only works because the EKF doesn't consume until the buffers have been full at least o [... cut]