ArduPilot/ardupilot #25327: Reports of drones breaching the geofence

czbxzm, 2023-10-20:
## Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[√] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

**Hardware type**
We verified the bug both in SITL simulation and on pixhawk

**Logs**
Two ways:
the first way:
Attack Steps:
- The operator intentionally controls the UAV to touch the edge of the electronic geo-fence to trigger a response to enter RTL mode
- When the drone enters RTL mode, it will fly to a preset return point.
- By setting the return point in a no-fly zone such as an airport, the aircraft will fly in the no-fly zone.
[image]

the second way:
https://github.com/ArduPilot/ardupilot/blob/c65675a64d453e28336230ac3bccad902b1133f3/libraries/AC_Fence/AC_Fence.cpp#L476
Because of this part of the code, when the values of both the was_breached and breached variables are true, the check for breaking the geofence will consistently return false, even if the drone is in a no-fly zone at the moment to perform its mission.
When a drone is in the air, paint the drone into a no-fly zone and the aircraft can still perform its mission.

Further exploitation of the second bug：
While the second option may seem less threatening, as it is the user breaking out of the no-fly zone they have drawn for themselves, further exploitation exists based on this vulnerability.
When the operator is malicious:
- Ardupilot supports importing no-fly zone files, and some diy aircraft manufacturers using this open source flight control choose to import airport no-fly zone information for their aircraft ahead of time.
- A malicious user could draw a new no-fly zone right next to the edge of an existing no-fly zone and close it.
- The new no-fly zone is turned on while the drone is flying in the new no-fly zone.
- There would be no additional status check at the junction of the two no-fly zones, which would allow a malicious operator to enter the existing airport no-fly zone.
[image]

amilcarlucas, 2023-10-20:
Which firmware version did you use to test this?

czbxzm, 2023-10-20:
谢谢，您的邮件梁子益已收到祝好!

czbxzm, 2023-10-22:
> Which firmware version did you use to test this?

the latest.

rmackay9, 2023-10-22:
If Dijkstra's/A-start object avoidance is enabled you may find it does better.

In general though, Auto, RTL, etc do not fly a path back to home which keeps the vehicle within the polygon fences.  The reason is that we worry the vehicle will get stuck as it tries to return.

rmackay9, 2023-10-22:
By the way, in general AP does not attempt to protect against "malicious operators".  The operator is assumed to have the best knowledge of what the vehicle should do and AP does not try to second guess that.

czbxzm, 2023-11-22:
> By the way, in general AP does not attempt to protect against "malicious operators". The operator is assumed to have the best knowledge of what the vehicle should do and AP does not try to second guess that.

Thank you for your response, sir.I agree with you, don't assume users are malicious. It is possible for a benign user to breach a no-fly zone without any warning. This could result in them accidentally flying into an airport breaking the law, or having the plane fly into the woods and crash.  

When the drone is flying normally in the flyable zone, it will not enter the no-fly zone anyway, which makes the user feel that the geo-fence is reliable and logical. But when the drone touches the fence and enters rtl mode, or activates the fence in the middle of a mission, it quietly fails.

In addition, the failure of a single new geofence may further affect more existing geofences, as I have already mentioned in "Further exploitation" above.

I think this problem arises because the logic of prioritising no-fly zones is a bit confusing.We should fix the highest priority for geofencing, and even if a user wants to take over a no-fly zone they should first explicitly operate to cancel it.

After I reported the similar issue to PX4, they added a new check today to verify the relationship between home points and no-fly zones, which fixed the RTL mode breaching geofence issue.And they have now forbidden the uploading of geofences while the aircraft is armming, effectively fixed “the second way” mentioned above.

https://github.com/PX4/PX4-Autopilot/pull/22394

czbxzm, 2023-11-22:
Sorry for closing this issue by my mistake!