ArduPilot/ardupilot #25381: AP_DDS: Add waypoint support

Ryanf55, 2023-10-27:
## Feature request

**Is your feature request related to a problem? Please describe.**

ROS users would like to send waypoint(s) from a companion computer to ArduPilot and have it follow the waypoints. 

**Describe the solution you'd like**

Follow REP-147, or another commonly adopted formats. Support optionally describing the orientation of the drone at each waypoint, or let the autopilot optimize orientation.

**Describe alternatives you've considered**

Mavlink currently supports this

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**

N/A

ohyaiamhere, 2024-01-29:
Hello Ryan.

I am a beginner and would like to take this up. Please let me know how I can help.

Ryanf55, 2024-02-28:
Correct

haarshitgarg, 2024-02-29:
I have added `set_global_location` functions similar to AP_ExternalControl_Plane for copter and rover and I have the build and SITL simulation environment configured. How can I test my changes before I create a pull request?

Ryanf55, 2024-02-29:
Awesome, great to hear. For testing, have you gone through these tutorials?
* https://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html
* https://ardupilot.org/dev/docs/ros2.html
* https://ardupilot.org/dev/docs/ros2-sitl.html

Once you have done all of those (in order), you can follow the instructions in the original PR for plane and repeat the process to check the Copter does the same behavior.
* https://github.com/ArduPilot/ardupilot/pull/25722#issue-2030071065

If you get stuck, please reach out. Also, if you can, please create a "draft" style pull request with your changes. I can start providing feedback that way. 

Many thanks for the contribution!

haarshitgarg, 2024-03-03:
Hello @Ryanf55 It took me some time to set up the testing environment. I tried the goal interface support for plan that you implemented it worked for me once but now I am getting this message
[image]
I am not sure what I am doing wrong.

haarshitgarg, 2024-03-03:
Hello @Ryanf55,  I have updated the PR and did the tests [here](https://github.com/ArduPilot/ardupilot/pull/26362).

Ryanf55, 2024-05-12:
This is now merged for copter, plane and rover.

closed 2024-05-12 by Ryanf55