ArduPilot/ardupilot #25382: AP_DDS: Add trajectory tracking control for Plane

Ryanf55, 2023-10-27:
## Feature request

**Is your feature request related to a problem? Please describe.**

The external control interface currently supports velocity control in body and earth frame. While #25381 is intended for high-level waypoint control, there is another desired form - trajectory control, that works well for plane.

**Describe the solution you'd like**

* Design/pick a ROS message for trajectory control, based on mavlink, REP-147, and current research.
* Add the interface for trajectory control in AP_ExternalControl
* Implement the trajectory control in Plane
* Write up a ROS 2 example demonstrating usage on a lemniscate or oval pattern
* Contribute documentation to the wiki on how to run it

**Describe alternatives you've considered**

Velocity control as a workaround

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

**Additional context**

https://ieeexplore.ieee.org/document/5611328

Astik-2002, 2024-02-09:
Hello Ryan. Is someone working on this issue actively? If not, I'd like to work on this issue.

Astik-2002, 2024-02-09:
I believe we can already control the position and yaw of a fixed-wing in guided mode. Do we need further inner-loop control with this?

Ryanf55, 2024-02-09:
Yea, I thought so too. Did you see this PR?
https://github.com/ArduPilot/ardupilot/pull/26143