ArduPilot/ardupilot #25433: Quadplane: Target speed logic in VTOL transition to cruise.

AndKe, 2023-11-01:
## Bug report

To reproduce: 

1 manual Take-off in Q-hover/loiter,  
2 keep RC throttle high, in order to climb.
3 at desired transition altitude, switch to Cruise
At this point, it is not intuitive to instantly lower throttle input, and keep throttle high as we wish to climb.
4 observe that during, and after the completed transition, the plane will do uncommanded descend. 

To me, this looks like the high throttle causes a very high demand for speed, which is met by sacrificing the altitude.
This can even be performed as the RC pilot inputs full throttle, and full up elevator( limited by fly-by-wire)

So the feeling is: a pilot takes off, and as terrain/obstacle approaches, it may be logical to input full throttle, and (given FBW) full up-elevator: and the result is a descent, at high speed.

[image]

**Version**
AP 4.4  (and older, and current release)
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadplane

**Hardware type**
CubeOrange

**Logs**
https://drive.google.com/file/d/1KpTksvhAd7ZrFNI5hlugjaATbU38_200/view?usp=sharing

IamPete1, 2023-11-01:
The TECS height demands are in the TECS log msg.

[image]

The input height from the pilot is in blue, the TECS target is in green. The problem is that the vehicle gains 7 or 8 meters in its transition, this puts is above its target height so it descends. Very soon after the pilot height demand increases the vehicle starts climbing again. It is only ever 1m under the altitude at which the transition finished.

IamPete1, 2023-11-01:
I also notice you have quite a large thrust to pitch coupling, this means that at high throttle the vehicle was quite out of trim when it first transitioned, this could be the cause of that climb.

[image]

The pitch I term follows the throttle level quite closely. You could try to minimize this with `KFF_THR2PTCH` or double check the physical thrust line on the vehicle.

AndKe, 2023-11-07:
@IamPete1  
you wrote: "Very soon after the pilot height demand increases the vehicle starts climbing again. It is only ever 1m under the altitude at which the transition finished."

Please see below: the altitude drop is 7 meters, and seems to start the moment I pull up.
The worst is the scenario:
I see IAS increase well beyond minimum speed and even cruise and the plane is descending : and this is while pilot is  pull up and at high throttle.
There was plenty of energy to respond properly to those inputs, and no need to pitch down.

[image]