ArduPilot/ardupilot #25434: Quadplane: Transition fails if blowing with the wind during VTOL climb

AndKe, 2023-11-01:
## Bug report

When starting in a significant headwind, it may be tempting to drift off backward, in order to do the transition closer to oneself. / or to save energy, rather han holding position during VTOL climb (pitching nose down towards headwind)

To reproduce: 
1 Climb in Qloiter or Qstabilize, keep nose into the wind, let the UAV drift "backward"
2 when at altitude - switch to (for example) Cruise
3 observe that the transition fails.

To me, it seems that the wind estimate goes bad when drifting/flying backward during VTOL climb.
[image]

**Version**
AP4.4.0  (and newer)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadplane

**Hardware type**
Cube orange

**Logs**
https://drive.google.com/file/d/1LPinfPQbLsfKwo_GzhEuuNPsuV78kXQ3/view?usp=sharing

IamPete1, 2023-11-01:
In this case you have a airspeed sensor so the wind estimate is not being used. 

It should not update when the VTOL motors are active. However we do get one change when you first switch into cruise:

[image]

@tridge  I think this is a race between `in_vtol_mode` going false and `in_assisted_flight` going true in `update_fly_forward`. The delay in changing from false to true we were talking about should fix this too.

In this case it looks like cruise locked in the flying backwards ground course, so it tried to do a 180 deg turn,  this is what I get in the sim:

[image]

So the fix is just that we are flying forward over the ground before letting cruise lock in heading.

IamPete1, 2023-11-01:
@AndKe can you check in FBWB to confirm if it is the attempt to do the 180 turn in cruise that is causing your issue.