ArduPilot/ardupilot #25435: EKF3 Accel  delVel bias bug

lixiaowei123, 2023-11-01:
[log.zip](https://github.com/ArduPilot/ardupilot/files/13226334/log.zip)

I have optimized the acceleration offset bug handled on the latest official website using Copter-4.0.7, but the following issues still occur
[image]

[image]

IamPete1, 2023-11-01:
Can you proved more detail and re-test on Copter 4.4.

lixiaowei123, 2023-11-02:
> Can you proved more detail and re-test on Copter 4.4.

This kind of problem doesn't happen every time, it's not like the latest version appears，There must have been a problem with the code, regardless of which version,

lixiaowei123, 2023-11-02:
> > Can you proved more detail and re-test on Copter 4.4.
> 
> This kind of problem doesn't happen every time, it's not like the latest version appears，There must have been a problem with the code, regardless of which version,

@priseborough

IamPete1, 2023-11-02:
Sorry, I still don't understand the issue your trying to report. 

Under what circumstances does it happen?
What happens?
What do you expect to happen?
What tests have you done to narrow it down?
Does it happen on more than one flight controller?
Does it happen in the simulator?

Your plots seem to be comparing accelerations and velocities from the EKF? What are they showing?

lixiaowei123, 2023-11-02:
> Sorry, I still don't understand the issue your trying to report.
>
> Under what circumstances does it happen? What happens? What do you expect to happen? What tests have you done to narrow it down? [...]

I have been working normally in the morning, but I just unlocked it in the afternoon and there was a surge。What I want to express is that there is a bug in the calculation of acceleration velocity increment deviation

lixiaowei123, 2023-11-02:
[image]
[image]

lixiaowei123, 2023-11-02:
In either case, the vertical acceleration in the geographic coordinate system should be -9.8. You can see in the log that the acceleration anomaly in the geographic coordinate system is caused by the offset of the acceleration velocity increment

lthall, 2023-11-02:
> I have optimized the acceleration offset bug handled on the latest official website using Copter-4.0.7, but the following issues still occur

> This kind of problem doesn't happen every time, it's not like the latest version appears，

It appears you are using your own custom version of the code and are not able to reproduce on master or 4.4. If you are able to produce logs and demonstrate the problem on 4.4 or master we can reopen the issue.

closed 2023-11-02 by lthall

lixiaowei123, 2023-11-03:
Is it necessary to go to the latest testing for every version with problems? What is the logic behind this

lthall, 2023-11-03:
There is no point trying to fix a problem that we have already fixed or doesn't exist in the latest version of the code. This issue list isn't for your custom version of the code or for old versions of the code.

lixiaowei123, 2023-11-03:
I have reviewed the latest bug fix and it does not fundamentally solve this problem

lixiaowei123, 2023-11-03:
[image]
[image]
[image]

I have fixed all these issues and verified multiple times. My version is already the same as here, and the latest version on the official website has not resolved this issue

lixiaowei123, 2023-11-03:
I only have a set of IMUs, and there won't be the bug above

lthall, 2023-11-03:
If you can clearly articulate the problem and show provide a replay log using 4.4 or master we can re-open the issue.