ArduPilot/ardupilot #25455: lua scripts sometimes do not start or re-start

tridge, 2023-11-03:
in current master we have a bug where lua scripts sometimes don't start. There is only one board I've seen where this happened a lot, the tlog for that case is here:
http://uav.tridgell.net/tmp/scripts_not_starting.tlog
that setup is a CubeOrange running this lua script:
https://github.com/ArduPilot/ardupilot/pull/25428
the same script on 4.4.x didn't show the problem (but as the bug is intermittent this doesn't necessarily mean it is not in 4.4.x)
some key information from debugging so far:
 - copying identical parameters and script to a different CubeOrange did not reproduce the issue despite hundreds of attempts
 - we do at least get to the point of calling lua_scripts::reschedule_script when the bug happens, but the script does not run

InnovativeUAVs, 2024-02-16:
In case there's interest in more info on this bug, I can confirm seeing a similar problem. I have several CubeOrange and CubeOrange+. At first I thought the problem was on me, but I have seen it several times now and have reproduced it in SITL. It seems to happen when I delete all the scripts, and then add new scripts, then restart the AP (but I could be wrong). I am doing this frequently as I'm testing for various failure modes in LUA scrips and their impact (or not) on the AP code/function. If there's interest, I'll formalize my test procedures and update my findings here.

rmackay9, 2024-02-17:
@InnovativeUAVs,

Thanks for that.  If you've got a set of steps that can reproduce the problem that would be a great help towards getting to the bottom of the issue.

IamPete1, 2024-03-07:
@InnovativeUAVs Any update? Steps to reproduce would be very useful.

InnovativeUAVs, 2024-03-08:
I have some additional scripts to add soon, and I will try again to reproduce. I spent some considerable time trying to reproduce before, and could not. But, I can confirm there is some condition wherein this is repeatable, and it seems to be related to first configuration / early setup work. Once my FCs are configured, I can't reproduce the error. I will continue to try and will update here if of note. Thanks for the help.

IamPete1, 2024-03-08:
Thanks for the update. The only thing I can add is that scripts will not run when you first set scripting enable. You will need a full flight controller restart, just a scripting restart is not enough. We could update the messages to make that more clear.