ArduPilot/ardupilot #25464: BICOPTER / VTOL CONFUSION

vgrsystems, 2023-11-05:
COPTER version V4.4.2
Pixracer R15
When configuring Bicpoter (Frame_class=10), Mission Planner report a VTOL-DUALROTOR in the top right and yaw is indeed acting like a vtol birotor.
[image]
Thanks a lot,
Regards,

IamPete1, 2023-11-06:
I don't understand the problem. A bicopter is a VTOL-dual rotor? It sounds like the vehicle is behaving correctly? Just the com port name is not as you would have expected?

vgrsystems, 2023-11-06:
Hi,
The bicopter is not a VTOL dual rotor. It's a birotor without fix wing and it belongs to the copter firmware. So the comport name seems wrong because the VTOL Duo rotor is a fixed wing and it belongs to arduplane firmware. 
Moreover, when I configure copter frame as a bicopter, there are FWBA and FWBB in the flight modes that are related to FW frame. It's why I suspect a confusion between FW VTOL and Birotor?
Txs,rgds

IamPete1, 2023-11-06:
If you update to beta mission planner you should get the correct mode names. I agree that VTOL is typically only used to refer to vehicles with wings but being capable of vertical takeoff and landing does not require wings

vgrsystems, 2023-11-06:
Great, Iwill do it,
Many thanks
Rgds

vgrsystems, 2023-11-06:
I updated mission planner with no luck, it stays with VTOL-DUOROTOR name of the comport anf flight modes didn't change. I noticed that some flight modes are in capital letter (they work) and others are in lower case (they don't work and show "no such mode").

IamPete1, 2023-11-07:
Sorry, which modes are you looking at? These work for me, its correctly showing the copter modes.

[image]

However you will spot that the mode in the HUD says manual, which is a plane mode and we get the plane icon on the map. 

Once I change modes the HUD mode name is correct and works normally.

vgrsystems, 2023-11-07:
I've got this one for my part
[image]
The flight modes in uppercase are not available. The HUD displays "no such mode". Maybe it's a problem of the 1MB firmware limitation. I will check with another pixracer and tell you. But do you have a plane icon on the map even when you switch the mode? For me, even when I switch to a copter flight mode, the plane icon doesn't change. Thanks, François

vgrsystems, 2023-11-10:
Pete,
I tried with different pixracer with the same results. The only available modes are those in capital letter. The others (loiter for example) are not available (no such mode in HUD). I will try to use QLOITER to see if it's ok but it's strange that I'm in quadplane mode and you in copter mode with the same FRAME_CLASS. Txs.

vgrsystems, 2023-11-12:
EDIT: 
USB Mission planner -> bug with wrong flight modes
USB Qgroundcontrol -> no bugs as long as usb connected
Radio telemetry Qgroundcontrol -> bug with wrong flight modes
Radio telemetry Mission planner -> bug with wrong flight modes
So it seems to depend on the link with the GCS software.