ArduPilot/ardupilot #25492: vehicle:get_target_location() in lua Copter

Alexandre-Larribau, 2023-11-08:
## Bug report

**Issue details**

I described everything in [this ](https://discuss.ardupilot.org/t/vehicle-get-target-location-in-lua-copter/108901/3)topic. I am available if you need more details but to resume it looks like the vehicle:get_target_location() parameter works fine in Plane but not in Copter in lua scripts. I share my test code in the issue.  

**Version**
I am on windows 11, WSL 2, Ubuntu 22.04

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter V4.5
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
default parameters SITL 

**Hardware type**
default parameters SITL 

**Logs**

robot-to-society, 2024-07-01:
@rmackay9 I think this issue can be closed.

peterbarker, 2024-08-03:
Thanks @robot-to-society - closing this one.

closed 2024-08-03 by peterbarker