ArduPilot/ardupilot #25493: Plane: Why does it fly in circles when it reaches the nav waypoint? Can you disable it on-demand?

Sefibrah, 2023-11-08:
## Feature request

**Is your feature request related to a problem? Please describe.**
In GUIDED mode I am trying to send a MAV_CMD_NAV_WAYPOINT command via pymavlink library in Python to my gazebo-zephyr model in SITL. It gets accepted, the plane goes to it, BUT instead of passing through it, the plane starts circling it?

**Describe the solution you'd like**
I want a way to disable this when I want. Or at least to make it go through it first, then it can circle it. As long as it passes through it once, I'll be fine. Some kind of parameter in the Parameter list that I can modify, or a parameter that I can modify directly from the MAV_CMD_NAV_WAYPOINT params list should be sufficient.

Or at least if someone can point me to the right direction to implement this solution myself, I'd gladly take it as well. This world is after all already busy enough with its own problems.

**Describe alternatives you've considered**
I tried WP_RADIUS, WP_MAX_RADIUS, and WP_LOITER_RAD, none of which worked...

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Additional context**
This is how it looks in the map:
[image]

This is how it looks in the console:
[image]

This is how it looks in the gazebo simulation:
[image]

This is the forum [link](https://discuss.ardupilot.org/t/why-does-sitl-simulator-only-fly-circles/99989) for even more clarification.

magicrub, 2023-11-08:
This is normal behavior. What did you expect your fixed-wing aircraft, that must fly forward to stay aloft, do once it reached your point?

For guided, just keep giving it a point that it can't reach like a carrot on a stick.

magicrub, 2023-11-08:
Proper use for GUIDED mode use is DO_REPOSITION. A normal list of waypoints is for AUTO mode

Sefibrah, 2023-11-09:
Was expecting a "passthrough-first-circle-second" behavior or something like it. Did it with DO_REPOSITION approach per your recommendation, but it still shows the same behavior:

[image]

[image]

Sefibrah, 2023-11-09:
You said:
> For guided, just keep giving it a point that it can't reach like a carrot on a stick.

What if I reallllyyyy want it to happen this way? The way I mentioned in my suggested solution I mean?

magicrub, 2023-11-09:
Only way to do that is in AUTO. Set a waypoint with acceptance = 1 followed by a loiter point on the same spot

Sefibrah, 2023-11-10:
> Only way to do that is in AUTO. Set a waypoint with acceptance = 1 followed by a loiter point on the same spot

Thank you for the kind explanations!

Ryanf55, 2024-12-30:
> Only way to do that is in AUTO. Set a waypoint with acceptance = 1 followed by a loiter point on the same spot

I'm re-opening this issue. I need to be able to do this from a companion computer feeding a list of dynamic waypoints to the autopilot. I can't fly auto because I need some of the GUIDED overrides.

Ryanf55, 2024-12-30:
I propose that a [MAV_CMD_DO_REPOSITION](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_REPOSITION) with param3 set to `NaN` or `0.0`, that plane will interpret it as a fly through.

timtuxworth, 2024-12-30:
In my plane follow script I achieve this by setting the waypoint radius p3 = 2.0 (i.e. 2 meters), this means it flies to within 2 m of the location. You can't set it to 1 as that has "special meaning", but I agree - this is a problem that needs a solution.

timtuxworth, 2024-12-30:
> I propose that a [MAV_CMD_DO_REPOSITION](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_REPOSITION) with param3 set to `NaN` or `0.0`, that plane will interpret it as a fly through. If it is not fed another waypoint by the time is passes through the configured `WP_LOITER_RAD`, then it will circle the point.

0 (zero), actually coded as <= 1, is already a special case - it means "use WP_LOITER_RAD". Maybe use the value 2 as the special case?