ArduPilot/ardupilot #25498: Problems with clang cross compiling ardupilot

Lqs66, 2023-11-09:
## Bug report

To implement clang cross-compile ardupilot, you need to specify for clang the header files and libraries to be used for arm-none-eabi-gcc compilation, for this I configured and compiled with the following command:
`CC=clang CXX=clang++ CXXFLAGS="-O0 -g --target=arm-none-eabi -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include/c++/10.2.1 -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include/c++/10.2.1/arm-none-eabi -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include/c++/10.2.1/backward -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/include -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/include-fixed -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include"  ./waf configure --board=CUAVv5`

The following error occurs:
> clang: fatal error: unknown argument: '-mno-thumb-interwork'

Thereupon I removed the -mno-thumb-interwork compilation option from CUAVv5_cache.py and continued compiling, and again the following error occurred:

> /opt/gcc-arm-none-eabi-10-2020-q4-major/bin/... /lib/gcc/arm-none-eabi/10.2.1/... /... /... /... /arm-none-eabi/include/c++/10.2.1/cstdio:98:11: fatal error: no member named 'FILE' in the global namespace using ::FILE;

I further modified CUAVv5_cache.py to include '--sysroot=/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/... /arm-none-eabi' was removed to continue compiling and ended up with the following error:

> fatal error: out-of-line definition of 'lseek' does not match any declaration in 'AP_Filesystem_FATFS' off_t AP_Filesystem_FATFS::lseek(int fileno, off_t position, int whence)

Obviously so far the cross compilation is failing.

**Platform**
[  ] All
[  ] AntennaTracker
[*] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Ryanf55, 2023-11-11:
> ## Bug report
> To implement clang cross-compile ardupilot, you need to specify for clang the header files and libraries to be used for arm-none-eabi-gcc compilation, for this I configured and compiled with the following command: `CC=clang CXX=clang++ CXXFLAGS="-O0 -g --target=arm-none-eabi -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include/c++/10.2.1 -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include/c++/10.2.1/arm-none-eabi -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include/c++/10.2.1/backward -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/include -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/include-fixed -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include" ./waf configure --board=CUAVv5`
> 
> The following error occurs:
> 
> > clang: fatal error: unknown argument: '-mno-thumb-interwork'
> 
> Thereupon I removed the -mno-thumb-interwork compilation option from CUAVv5_cache.py and continued compiling, and again the following error occurred:
> 
> > /opt/gcc-arm-none-eabi-10-2020-q4-major/bin/... /lib/gcc/arm-none-eabi/10.2.1/... /... /... /... /arm-none-eabi/include/c++/10.2.1/cstdio:98:11: fatal error: no member named 'FILE' in the global namespace using ::FILE;
> 
> I further modified CUAVv5_cache.py to include '--sysroot=/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/... /arm-none-eabi' was removed to continue compiling and ended up with the following error:
> 
> > fatal error: out-of-line definition of 'lseek' does not match any declaration in 'AP_Filesystem_FATFS' off_t AP_Filesystem_FATFS::lseek(int fileno, off_t position, int whence)
> 
> I changed "off_t AP_Filesystem_FATFS::lseek(int fileno, off_t position, int whence)" to "int32_t AP_Filesystem_FATFS::lseek(int fileno, int32_t position, int whence)". FATFS::lseek(int fileno, int32_t position, int whence)" to "int32_t AP_Filesystem_FATFS::lseek(int fileno, int32_t position, int whence)", you can continue to compile until the following error occurs:
> 
> > [581/886] Compiling libraries/AP_HAL_ChibiOS/SoftSigReaderInt.cpp
> > clang: warning: optimization flag '-fsingle-precision-constant' is not supported [-Wignored-optimization-argument]
> > clang: warning: optimization flag '-fno-strength-reduce' is not supported [-Wignored-optimization-argument]
> > clang: warning: argument unused during compilation: '-u _printf_float' [-Wunused-command-line-argument]
> > clang: warning: argument unused during compilation: '--specs=nano.specs' [-Wunused-command-line-argument]
> > clang: warning: argument unused during compilation: '--specs=nosys.specs' [-Wunused-command-line-argument]
> > warning: unknown warning option '-Wlogical-op'; did you mean '-Wlong-long'? [-Wunknown-warning-option]
> > warning: unknown warning option '-Werror=unused-but-set-variable'; did you mean '-Werror=unused-const-variable'? [-Wunknown-warning-option]
> > In file included from ../../libraries/AP_HAL_ChibiOS/UARTDriver.cpp:17:
> > In file included from ../../libraries/AP_HAL/AP_HAL.h:8:
> > In file included from ../../libraries/AP_HAL/AP_HAL_Main.h:19:
> > In file included from ../../libraries/AP_HAL/HAL.h:18:
> > ../../libraries/AP_HAL/CANIface.h:29:1: warning: 'CANFrame' defined as a struct here but previously declared as a class; this is valid, but may result in linker errors under the Microsoft C++ ABI [-Wmismatched-tags]
> > struct AP_HAL::CANFrame {
> > ^
> > ../../libraries/AP_HAL/AP_HAL_Namespace.h:41:5: note: did you mean struct here?
> > class CANFrame;
> > ^~~~~
> > struct
> > In file included from ../../libraries/AP_HAL_ChibiOS/UARTDriver.cpp:26:
> > In file included from ../../libraries/AP_Math/AP_Math.h:13:
> > In file included from ../../libraries/AP_Math/matrix3.h:43:
> > ../../libraries/AP_Math/vector2.h:42:1: warning: 'Vector2' defined as a struct template here but previously declared as a class template; this is valid, but may result in linker errors under the Microsoft C++ ABI [-Wmismatched-tags]
> > struct Vector2
> > ^
> > ../../libraries/AP_Math/vector3.h:69:1: note: did you mean struct here?
> > class Vector2;
> > ^~~~~
> > struct
> > ../../libraries/AP_HAL_ChibiOS/UARTDriver.cpp:93:1: warning: field 'serial_num' will be initialized after field 'sdef' [-Wreorder-ctor]
> > serial_num(_serial_num),
> > ^
> > ../../libraries/AP_HAL_ChibiOS/UARTDriver.cpp:108:30: warning: unknown warning group '-Wframe-larger-than=128', ignored [-Wunknown-warning-option]
> > #pragma GCC diagnostic error "-Wframe-larger-than=128"
> > ^
> > ../../libraries/AP_HAL_ChibiOS/UARTDriver.cpp:466:9: **fatal error: use of undeclared identifier 'vprintf_console_hook'
> > vprintf_console_hook = hal_console_vprintf;**
> > ^
> > 6 warnings and 1 error generated.
> 
> Obviously so far the cross compilation is failing.
> 
> **Platform** [ ] All [ ] AntennaTracker [*] Copter [ ] Plane [ ] Rover [ ] Submarine

What are you cross compiling for, and why are you not using waf?

Lqs66, 2023-11-11:
> CC=clang CXX=clang++ CXXFLAGS="-O0 -g --target=arm-none-eabi -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include/c++/10.2.1 [...]

In fact, that's what I'm using, and I'm using `CC=clang CXX=clang++ CXXFLAGS="-O0 -g --target=arm-none-eabi -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include/c++/10.2.1 -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include/c++/10.2.1/arm-none-eabi -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include/c++/10.2.1/backward -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/include -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/include-fixed -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include"  ./waf configure --board=CUAVv5` to configure it.
I chose clang for compilation because I needed to use llvm's static analysis capabilities.

Ryanf55, 2023-11-11:
> > CC=clang CXX=clang++ CXXFLAGS="-O0 -g --target=arm-none-eabi -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include/c++/10.2.1 -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include/c++/10.2.1/arm-none-eabi -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include/c++/10.2.1/backward -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/include -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/include-fixed -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include" ./waf configure --board=CUAVv5
> 
> I actually use waf and I use "CC=clang CXX=clang++ CXXFLAGS="-O0 -g --target=arm-none-eabi -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include/c++/10.2.1 -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include/c++/10.2.1/arm-none-eabi -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include/c++/10.2.1/backward -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/include -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/include-fixed -I/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/include" ./waf configure --board=CUAVv5" to configure it. I chose clang for compilation because I needed to use llvm's static analysis capabilities.

I see that you have cross posted between the forums, here, and discourse. 
[image]

Please avoid spamming all the places for support; let's follow up in the discourse thread.

Ryanf55, 2023-11-11:
Tagging this for wiki needed. The wiki instructions for building on Ubuntu 22 with clang do not work. It's missing packages from the default install like clang. It's not specific to cross-compile, it doesn't work even on native compilation following BUILD.md.

Lqs66, 2023-11-11:
> I see that you have cross posted between the forums, here, and discourse. [image]
>
> https://discuss.ardupilot.org/t/problems-with-compilation/108985/8
>
> Please avoid spamming all the places for [...]

Sorry, I'll undo what I did on discord.

khancyr, 2023-11-11:
Clang compilation is only working for SITL and Linux build.

If you just want static analysis, just build SITL, beside of the os layer, the vehicle code is the same so you will be able to do the analysis you want.

Lqs66, 2023-11-11:
> Clang compilation is only working for SITL and Linux build.
> 
> If you just want static analysis, just build SITL, beside of the os layer, the vehicle code is the same so you will be able to do the analysis you want.

Still, I would like to know the reason for this error.
`…/…/libraries/AP_HAL_ChibiOS/UARTDriver.cpp:466:9: fatal error: use of undeclared identifier ‘vprintf_console_hook’
vprintf_console_hook = hal_console_vprintf;
^
6 warnings and 1 error generated. `

Lqs66, 2023-11-15:
> Clang compilation is only working for SITL and Linux build.
> 
> If you just want static analysis, just build SITL, beside of the os layer, the vehicle code is the same so you will be able to do the analysis you want.

By "linux build" do you mean the navio2 board? Can clang be used to cross-compile raspberry pi + navio2?

828sky, 2024-09-24:
I have been failing for 4 days, long hours, can't figure this out. So many errors in compiling. Now at this one. If we want more than 3 temp sensors we have to figure this out to enable the AP macro.

Lqs66, 2024-09-24:
> I have been failing for 4 days, long hours, can't figure this out. So many errors in compiling. Now at this one. If we want more than 3 temp sensors we have to figure this out to enable the AP macro.

Indeed clang can only be used for linux builds such as raspberry pi etc. If you want to build non-sitl I recommend navio2.

faddey-w, 2025-01-23:
After running `Tools/environment_install/install-prereqs-mac.sh` I could compile ArduPlane with this command: `CXX=/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/arm-none-eabi-g++ CC=/opt/gcc-arm-none-eabi-10-2020-q4-major/bin/arm-none-eabi-gcc ./waf configure --board SpeedyBeeF405WING`

To find the compiler paths  - look into outputs of `install-prereqs-mac.sh`

closed 2025-05-11 by peterbarker