ArduPilot/ardupilot #25508: H743 I2C bug

geofrancis, 2023-11-10:
I have been upgrading a boat from a Omnibusf4pro to a Matek H743 Wlite. but I have ran into an issue with the i2c bus where setting an address of 0x70 on my sr04 sonar adapter (default for maxbotix i2c sonar) causing the i2c bus to lock up and stop detecting devices. 

the sonar was working on the omnibusf4pro before I changed over the controller so I thought I might have damaged something.
 I originally thought it was a wiring issue, so tested the adapter on a different H743 and it had the same result
 I thought the atmega168 might have got damaged somehow, so I swapped it for another 168 and same result.
 I thought it could be an issue with the 168 so I changed it for a 328 and same result 
so just to rule out some kind of software problem, I connected yuris i2c sensor adapter that runs on a 168 with address 9 and it worked fine. 

so i went and got the original 168 and tested it again with the omnibusf4pro and it works so now i know it's definately not a sensor or software or Arduino issue.

 I tried changing the sonar adapter to the same 0x09 9 address as the sensor adapter and it shows up on the i2c scanner

if i change the sonar address back to its default 0x70 112 without doing anything else it will show up for a few seconds then the i2c bus stops. 

[image]

rmackay9, 2023-11-10:
Thanks for the investigation and report.

I wonder if it could be that the autopilot has another I2C device that is also using 70?

geofrancis, 2023-11-11:
> Thanks for the investigation and report.
> 
> I wonder if it could be that the autopilot has another I2C device that is also using 70?

as far as I know, the only other i2c device on the board is the barometer and its on i2c bus 2.

Black6spdZ, 2024-06-14:
any update? seeing this exact same issue

geofrancis, 2024-06-14:
@rmackay9 I think i have found the problem, the lightware and maxbotix i2c drivers are hard coded to i2c 1 since thats the external i2c bus on a pixhawk. That is why I didnt have any issues when I first made the adapter as I was using a pixhawk1 but most f4 and h7 controllers use i2c 1 as internal and i2c 0 as external.

https://github.com/ArduPilot/ardupilot/pull/6260

geofrancis, 2024-06-25:
@rmackay9 I think something is wrong with the maxbotix driver and thats causing the problems with the i2c bus, do you have a maxbotix i2c sonar you could test with?