ArduPilot/ardupilot #25516: ros2 foxy cannot set for Copter 4.3

ZhongmouLi, 2023-11-11:
## Bug report
Branch  Copter 4.3 does not have Tools/ros2/ to configurate ROS2 foxy.

**Issue details**
1. tutorial at https://ardupilot.org/dev/docs/ros2.html downloads the whole ardupilot AGAIN at its main branch. 
2.  Copter 4.3 does not have Tools/ros2 for ros2 foxy configuratoin.

**Version**
Copter 4.3

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
Ubuntu 20.04
ROS2 foxy

**Logs**
_Please provide a link to any relevant logs that show the issue_

ZhongmouLi, 2023-11-11:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ardupilot_sitl: Cannot locate rosdep definition for [micro_ros_agent]
ardupilot_dds_tests: Cannot locate rosdep definition for [launch_pytest]
```

amilcarlucas, 2023-11-11:
The copter-4.3 branch doe not support ROS2. This is not a bug.

closed 2023-11-11 by amilcarlucas