ArduPilot/ardupilot #25517: AP_GPS_GSOF: Support GPS_RAW_DATA

Ryanf55, 2023-11-11:
## Feature request

**Is your feature request related to a problem? Please describe.**

Right now, there is no way to enable the internal logging on the PX1 GNSS  because the API is only available over the ethernet web UI.

**Describe the solution you'd like**

* Consume [GPS_RAW_DATA](https://ardupilot.org/copter/docs/parameters.html#gps-raw-data-raw-data-logging) in AP_GPS_GSOF
* Send enable logging/disable logging commands as appropriate
* Stretch: Handle runtime param updates

**Describe alternatives you've considered**

Connect laptop temporarily.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

abhinavs1920, 2023-11-12:
I would like to work on this issue. Please assign it to me.

abhinavs1920, 2023-11-14:
Thanks for assigning this issue to me. I am working on it.

Ryanf55, 2023-11-14:
> Thanks for assigning this issue to me. I am working on it.

Great. Let me know if you get stuck or are ready for a review.

abhinavs1920, 2023-11-14:
Can you please provide me with some references or contexts about the AP_GPS library?

Ryanf55, 2023-11-14:
The wiki is very brief: 
https://ardupilot.org/dev/docs/apmcopter-programming-libraries.html

The main interface for GPS is here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS.h

This library follows the frontend-backend split. 
https://ardupilot.org/dev/docs/code-overview-sensor-drivers.html#frontend-backend-split

Implementation that all GPS's follow is here:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS.cpp

GSOF header file: 
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS_GSOF.h

GSOF implementation: 
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_GPS/AP_GPS_GSOF.cpp

For retrieving a parameter from the param server, you can do it like so:
https://github.com/ArduPilot/ardupilot/blob/2e5af08a102d10b8b7d54414c7aa4f12f4a80ab9/libraries/AP_GPS/AP_GPS_GSOF.cpp#L63

Then, you need to write a validation function to ensure the user has not put a bogus value for the param, since param limits are not enforced:
https://github.com/ArduPilot/ardupilot/blob/2e5af08a102d10b8b7d54414c7aa4f12f4a80ab9/libraries/AP_GPS/AP_GPS_GSOF.cpp#L363
For example, if a user sets `GPS_RAW_DATA` to a value of `99`, the validation helper should return false.

Next, write the implementation of the logger configuration message, similar to `requestBaud`:
https://github.com/ArduPilot/ardupilot/blob/2e5af08a102d10b8b7d54414c7aa4f12f4a80ab9/libraries/AP_GPS/AP_GPS_GSOF.cpp#L164

The packet buffer structure was detailed on the initial issue description; seems like you can just hard code it as a `constexpr` buffer.

Make sure to calculate the checksum AFTER populating the transaction number `buffer[4]`.

kannishk, 2024-01-05:
Is this still open?

Ryanf55, 2024-01-05:
> Is this still open?

Yep!

shrey0303, 2024-01-06:
Hello, I was working on this issue, so I think the solution should follow as:

-> Get the GPS_RAW_DATA and validate as you mentioned
-> Then to handle enable/disable logging upon arm/disarm by fetching param1, if param1 = 1{AP_GPS_GSOF::requestBaud(const HW_Port portindex) and hard code as constexpr buffer and calculate checksum} else it writes nothing
Is it possible to directly add data log on flight controller memory using AP::logger().Write(Name,label......) instead of requestBaud function? 
As I am willing to contribute but I am new to this project, plz provide the necessary feedback and guidance

pulak-gautam, 2024-01-07:
@Ryanf55  The second last hexadecimal is supposed to be checksum right? Since 0x3 is an ending tag following how binary commands were handled in ``AP_GPS_GSOF::requestBaud``
but, hexadecimals preceding don't add up to that in the preceeding binary commands given 
for eg. Start Logging: 02 00 4C 09 08 44 45 46 41 55 4C 54 00 62 03  0x62 is csum but preceeding hex add up to 0x264. Could you please guide me as to where I am going wrong here?

Ryanf55, 2024-01-07:
> @Ryanf55 The second last hexadecimal is supposed to be checksum right? Since 0x3 is an ending tag following how binary commands were handled in `AP_GPS_GSOF::requestBaud` but, hexadecimals preceding [...]

The checksum is a uint8_t. We rely on natural overflow of the uint8_t for it to work. 0x264 larger than the max value of uint8_t. Also, the first 0x02 is also excluded.

pulak-gautam, 2024-01-19:
@Ryanf55 I have mostly done the work described in this ticket. But, I have been stuck in implementing the arming/disarming scenarios. I tried scanning through the source code for doing this but still couldn't find the right method to call. 

Could you help me out here, and point me in the right direction? I( am sorry for the delay, got involved in university work)

Ryanf55, 2024-01-20:
> @Ryanf55 I have mostly done the work described in this ticket. But, I have been stuck in implementing the arming/disarming scenarios. I tried scanning through the source code for doing this but [...]

Ah no worries, want to put what you have up for review in draft PR without the integration to arm/disarm? That's still quite useful, and could be merged after a review.

pulak-gautam, 2024-01-20:
> Ah no worries, want to put what you have up for review in a pull request without the integration to arm/disarm? That's still quite useful, and could be merged after a review.
Yes, will do it by the latest, after some cleanup. Thanks

pulak-gautam, 2024-01-21:
@Ryanf55 I have put up what I did in the pull request #26042. Thanks

Theonewhomadethings, 2024-03-26:
Is there still room to contribute on this issue? @Ryanf55

Saloni-2005, 2025-01-04:
@Ryanf55 Hello Sir I want to contribute in GSoC through  your organization and needed the guidance for the same.