ArduPilot/ardupilot #25520: Change the way horizon line is drawned on OSD when banked at 90 degrees to conform with aviation standards

dxs94, 2023-11-12:
## Feature request

Please note this is different from issue #24855 which was fixed in master.

**Is your feature request related to a problem? Please describe.**
When the aircraft is banked at 90°, the OSD artificial horizon line is not drawn the way it should be according to aviation standards. In this situation (90° of bank), the OSD no longer gives any indication of current pitch attitude. The drawing is the same no matter if the nose is below, on, or above horizon.

**Describe the solution you'd like**
Modify the way OSD horizon line is drawn.

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Additional context**

When bank is close to 0 horizon line is correct:

[image]

----------------------------

Now at bank 90°, the OSD line is always the same. No way of knowing if aircraft nose is below, on, or above horizon.

[image]

[image]

[image]

----------------------------

When bank is not 0, the horizon line should be drawn sideways to give pitch information. Here is a model of how it should look like:

[image]

Hwurzburg, 2023-11-24:
osd/ahrs uses euler angles...pitch becomes undefined at 90 roll....euler angle limitation...roll is undefined at +/-90 deg pitch conversely
either one becomes yaw
this is not a rotating ball like an aviation gyro....

dxs94, 2023-11-26:
Yes I'm the one who opened the issue regarding pitch which led to PR #5497. When I test flew it I realized there was room for further improvement.

Here I'm considering a further improvement to OSD artificial horizon which could be incorporated at the same time in upcoming AP 4.5. I'm aware of euler limitations, and at 90° bank, pitch is indeed controlled by yaw input. However please consider this: aircraft pitch body attitude is still defined at 90° bank, and this is what should be presented to the pilot (ie. drawn on OSD). Just how it is in mission planner: in below screenshots the pitch value as known to Ardupilot is reported, and the artificial horizon drawing is correct, unlike on OSD.

Level flight, roll 0:
[image]

-----------------
Level flight, roll 90°: the artificial horizon line is in the middle of the attitude display frame:
[image]

------------------
Aircraft in a climb, right roll 90°: the artificial horizon line is drawn on the RIGHT of the attitude display frame:
[image]

--------------------
Aircraft in a descent, right roll 90°: the artificial horizon line is drawn on the LEFT of the attitude display frame:
[image]

This obviously is reversed for a LEFT 90° roll.

Hwurzburg, 2023-11-26:
without major code changes 90 deg roll will have gimbal locked pitch, as will 90 deg pitch have gimbal locked roll...not something I would anticipate happening anytime soon, if ever

peterbarker, 2025-05-11:
> without major code changes 90 deg roll will have gimbal locked pitch, as will 90 deg pitch have gimbal locked roll...not something I would anticipate happening anytime soon, if ever

The AHRS is quite happy to hand you back a quaternion.  I guess the problems lie in the OSD code, then?