ArduPilot/ardupilot #25535: AP_DDS: MicroROS agent doesn't publish topics when GPS automatically gets GPS lock.

KyleJewiss, 2023-11-14:
## Bug report
AP_DDS: MicroROS agent doesn't publish topics when GPS automatically gets GPS lock.

We have Ardupilot running on a PX4 with a CubeOrange and a Here3+ GPS. We followed the documentation in the AP_DDS library to setup DDS/MicroROS: https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_DDS

We got it all working, where the data from the Here3+ could be published as topics to a ROS domain, however this is only the case when the Here3+ starts in a flashing blue "Disarmed with no GPS lock" state. When it is outside it automatically achieves GPS lock and flashes green "Disarmed (ready to arm), GPS lock acquired". See documentation here: https://docs.cubepilot.org/user-guides/here-3/here-3-manual

When the GPS starts without GPS lock the console output looks like the following:
[image]

When the GPS starts with GPS lock the console output looks like this:
[image]

**Version**
4.4 dev version

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X ] Rover
[  ] Submarine

**Hardware type**
Pixhawk4 with CubeOrange and a Here3+

**Logs**
[image]
[image][image]

russkel, 2023-11-14:
If you could add the mavlink log output as well that would be great. From memory, there were no mentions of AP_DDS showing in the log if the AP initialised with a hot GPS lock.

I had a look through the code and could not see any conditionals, especially anything that would prevent the output such as "DDS initialised" - could something be crashing?

KyleJewiss, 2023-11-15:
So I have connected to the cubeorange using the mavlink terminal console. This is the output when the device instantly find geofix:
[image]

This is the output when it doesn't. As you can see it calls the  AP_DDS this time:
[image]

Ryanf55, 2023-11-16:
Thanks for the detailed report. I have most of the similar hardware, I will try to reproduce it. Sorry for the trouble!

Ryanf55, 2023-11-16:
Which commit hash are you currently testing?

KyleJewiss, 2023-11-16:
Hey @Ryanf55, so I have narrowed the issue down to line 724 in the AP_DDS_client.cpp file.

[image]

The mavlink before the while loop on 724 is posted to the console, however the one inside the loop and after is not. This leads me to believe it is getting stuck in  the "uxr_create_session" which is from session.c.

KyleJewiss, 2023-11-16:
Hi,

Sorry it appears that it actually gets stuck earlier in the code but I had it commented out and forgotten about it:

```cpp
// check ping
        const uint64_t ping_timeout_ms{1000};
        const uint8_t ping_max_attempts{10};
        if (!uxr_ping_agent_attempts(comm, ping_timeout_ms, ping_max_attempts)) {
            GCS_SEND_TEXT(MAV_SEVERITY_ERROR, "DDS Client: No ping response, exiting");
            // return;
        }
```

The return is commented out because otherwise the hardware will stop trying to connect to the DDS client after 10 seconds. This has no bearing on the current issue. The code appears to get stuck in "uxr_ping_agent_attempts". We are theorising that it could be something to do with the locking transport stuff within the "uxr_ping_agent_attempts" function.

KyleJewiss, 2023-11-16:
I dont know exactly. That code should still execute regardless. I'll test a bit later

russkel, 2023-11-16:
> > Hi,
> > Sorry it appears that it actually gets stuck earlier in the AP_DDS_client.cpp code but I had it commented out and forgotten about it:
> > ```c++
> > // check ping
> >         const uint64_t ping_timeout_ms{1000};
> >         const uint8_t ping_max_attempts{10};
> >         if (!uxr_ping_agent_attempts(comm, ping_timeout_ms, ping_max_attempts)) {
> >             GCS_SEND_TEXT(MAV_SEVERITY_ERROR, "DDS Client: No ping response, exiting");
> >             // return;
> >         }
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > The return is commented out because otherwise the hardware will stop trying to connect to the DDS client after 10 seconds. This has no bearing on the current issue. The code appears to get stuck in "uxr_ping_agent_attempts". We are theorising that it could be something to do with the locking transport stuff within the "uxr_ping_agent_attempts" function.
> 
> Is there any way to reproduce this in SITL?

It works fine in SITL when configured normally, we are unsure how to simulate a cold GPS fix with SITL.

Ryanf55, 2023-11-17:
> It works fine in SITL when configured normally, we are unsure how to simulate a cold GPS fix with SITL.

My only suspicion is that time is frozen on the FC, and the loops don't advance on device without GPS. I worked with CubePilot last night on setting up the debugger to reproduce locally on my Cube Orange but ran out of time. I've seen it in https://github.com/eProsima/Micro-XRCE-DDS-Client/issues/350

KyleJewiss, 2023-11-17:
> > > > Hi,
> > > > Sorry it appears that it actually gets stuck earlier in the AP_DDS_client.cpp code but I had it commented out and forgotten about it:
> > > > ```c++
> > > > // check ping
> > > >         const uint64_t ping_timeout_ms{1000};
> > > >         const uint8_t ping_max_attempts{10};
> > > >         if (!uxr_ping_agent_attempts(comm, ping_timeout_ms, ping_max_attempts)) {
> > > >             GCS_SEND_TEXT(MAV_SEVERITY_ERROR, "DDS Client: No ping response, exiting");
> > > >             // return;
> > > >         }
> > > > ```
> > > > 
> > > > 
> > > >     
> > > >       
> > > >     
> > > > 
> > > >       
> > > >     
> > > > 
> > > >     
> > > >   
> > > > The return is commented out because otherwise the hardware will stop trying to connect to the DDS client after 10 seconds. This has no bearing on the current issue. The code appears to get stuck in "uxr_ping_agent_attempts". We are theorising that it could be something to do with the locking transport stuff within the "uxr_ping_agent_attempts" function.
> > > 
> > > 
> > > Is there any way to reproduce this in SITL?
> > 
> > 
> > It works fine in SITL when configured normally, we are unsure how to simulate a cold GPS fix with SITL.
> 
> My only suspicion is that time is frozen on the FC, and the loops don't advance on device without GPS. I worked with CubePilot last night on setting up the debugger to reproduce locally on my Cube Orange but ran out of time. I've seen it in [eProsima/Micro-XRCE-DDS-Client#350](https://github.com/eProsima/Micro-XRCE-DDS-Client/issues/350)

Okay we will look into that, thankyou for putting your time into this issue!

Ryanf55, 2023-11-18:
I've tried reproducing on 6X, and fail to get serial working. 6X has nothing connected except USB. 

```
./waf configure --enable-dds --board Pixhawk6X
./waf plane --upload

# parameters for mavproxy
DDS_ENABLE 1
SERIAL1_BAUD     115
SERIAL1_PROTOCOL 45
```

AND
```
ros2 run micro_ros_agent micro_ros_agent serial -b 115200  -r dds_xrce_profile.xml -D /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02
```

Result in autopilot:
```
AP: DDS Client: No ping response, exiting

```

And, no apparent data in the micro ros agent:
```
$ ros2 run micro_ros_agent micro_ros_agent serial -b 115200  -r dds_xrce_profile.xml -D /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02
[1700291849.324354] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1700291849.324839] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1700291872.071255] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1700291872.081351] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02, error 2, waiting for connection...
[1700291873.089339] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02, error 2, waiting for connection...
[1700291874.097161] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02, error 2, waiting for connection...
[1700291875.104609] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02, error 2, waiting for connection...
[1700291876.112400] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02, error 2, waiting for connection...
[1700291877.119291] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02, error 2, waiting for connection...
[1700291877.860837] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1700292086.799159] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1700292086.809247] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02, error 2, waiting for connection...
[1700292087.817031] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02, error 2, waiting for connection...
[1700292088.824711] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02, error 2, waiting for connection...
[1700292089.832190] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02, error 2, waiting for connection...
[1700292090.839454] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02, error 2, waiting for connection...
[1700292091.846572] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02, error 2, waiting for connection...
[1700292092.658736] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3

```

srmainwaring, 2023-11-20:
@Ryanf55 running tests on MatekH743-WING (v1) installed into a tilt-rotor VTOL.

Vehicle on bench, has

- USB attached for console on `SERIAL0` mapped to `/dev/cu.usbmodem142401`
- SiK telem radio on `SERIAL2` (TX1/RX1) mapped to `/dev/cu.usbserial-0001`
- CP2102 TTL - USB on `SERIAL5` (TX8/RX8) mapped to `/dev/cu.usbserial-4`

Config and install:

```bash
./waf configure --board MatelH743 --enable-dds
./waf plane --upload
```

#### TTL-USB connection

```bash
SERIAL2_BAUD     57          # 57600
SERIAL2_PROTOCOL 2.0         # MAVLink2
SERIAL5_BAUD     57          # 57600
SERIAL5_PROTOCOL 45.0
```

```bash
./install/micro_ros_agent/lib/micro_ros_agent/micro_ros_agent serial --dev /dev/cu.usbserial-4 --baudrate 57600 --verbose 6 --refs /Users/rhys/Code/ros2/humble/ros2-ardupilot/install/ardupilot_sitl/share/ardupilot_sitl/config/dds_xrce_profile.xml
```

Connection successful.

#### SiK telem connection

Flip the roles of `SERIAL2` and `SERIAL5`:

```bash1
SERIAL2_BAUD     57          # 57600
SERIAL2_PROTOCOL 45.0
SERIAL5_BAUD     57          # 57600
SERIAL5_PROTOCOL 2.0         # MAVLink2
```

```bash
./install/micro_ros_agent/lib/micro_ros_agent/micro_ros_agent serial --dev /dev/cu.usbserial-0001 --baudrate 57600 --verbose 6 --refs /Users/rhys/Code/ros2/humble/ros2-ardupilot/install/ardupilot_sitl/share/ardupilot_sitl/config/dds_xrce_profile.xml
```

Connection fails.

I can see the initial ping message. Run `screen` then reboot the autopilot from mavproxy: 

```bash
$ screen /dev/cu.usbserial-0001 57600

~�
 ����
    ���~�
        ����~�
             ����
                ���~�
                    ����~�
                         ����
                            ���
```

I suspect there may be a timeout in the second case.

Ryanf55, 2023-11-20:
I did some more testing. Despite having SERIAL1_PROTOCOL set to 45, I still see some GCS status message texts coming out the port. Even if DDS_ENABLE to 0, there is still data being shoved out the port.

[image]

Mavlink is hogging the port:
```
$ mavproxy.py --master /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02 
Connect /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02 source_system=255
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from /dev/serial/by-id/usb-ArduPilot_Pixhawk6X_210028000151323131373139-if02
MAV> Detected vehicle 1:1 on link 0
online system 1
MANUAL> Mode MANUAL
AP: ArduPlane V4.5.0-dev (d625a1b1)
AP: ChibiOS: 3ef1657d
AP: Pixhawk6X 00280021 31325101 39313731
AP: IOMCU: 410 2003 412FC231
AP: RCOut: PWM:1-16
AP: IMU0: fast sampling enabled 2.0kHz
Received 1005 parameters (ftp)
Saved 1005 parameters to mav.parm
Flight battery warning

```

Ryanf55, 2023-11-20:
Can you repeat the same test as Rhys and I?

1. Run `master` firmware on your Cube orange `./waf configure --board CubeOrange --enable-dds` and `./waf copter --upload`
2. Set the following params to enable DDS at 57K (try this instead of 115k for now), and replace `x` number with the serial port you are using for DDS serial traffic on your Cube.
   * SERIALx_BAUD     57          # 57600
    * SERIALx_PROTOCOL 45.0
 3. Run the hex dump on 57k, replacing `/dev/ttyX` with the serial port that your companion computer uses to talk ROS 2. 
 ```
 python3 -m serial.tools.miniterm /dev/ttyX 57600 --echo --encoding hexlify
 ```
 
 Paste the output of 3. If it's what we expect, there should be 10 ping attemps. What I see is Mavlink stomping on the serial traffic and interleaving data which corrupts the stream.

KyleJewiss, 2023-11-20:
Thanks for all the work on this. Yep, I'll do this and get back to you in 2 hours or so.

srmainwaring, 2023-11-20:
@KyleJewiss, if you have a TTL to USB serial adapter another test would be to use a different UART (not one of the ones usually assigned to Telem or GPS) for the DDS data. That may give you a temporary workaround while we determine what is causing the issue Ryan is observing with the USB attached to SERIAL1.

Ryanf55, 2023-11-20:
Tagged this for dev call to ask the team if there is any known reason that SERIAL1_PROTOCOL would be ignored.

KyleJewiss, 2023-11-20:
Alright, so I rebased to master and I tested both of the serial ports available on the CubeOrange, and when the here3+ gets a hot geofix there is NOTHING at all coming across the DDS Serial port. 

```
➜  ardupilot git:(gama) ✗ python3 -m serial.tools.miniterm /dev/ttyUSB0 57600 --echo --encoding hexlify
--- Miniterm on /dev/ttyUSB0  57600,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

```

When it doesn't get hot geofix it does.

```
python3 -m serial.tools.miniterm /dev/ttyUSB0 57600 --echo --encoding hexlify       
--- Miniterm on /dev/ttyUSB0  57600,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AA AA BB BB 81 00 FC 01 3D 01 7E 00 00 18 00 80 00 00 00 00 01 10 00 58 
````

We have figured out a work around for now. We have added this 5 second delay in ```AP_DDS_Client.cpp``` main_loop.
```
void AP_DDS_Client::main_loop(void)
{
    // Work around to stop DDS & mavlink conflict
    hal.scheduler->delay(5000);

    if (!init_transport()) {
        return;
    }
```
This seems to avoid the mavlink/transport conflict.

Ryanf55, 2023-11-20:
On CubeOrange, serial0 and serial6. The first and last are USB. Check the hwdef.daf (see otg and otg2). 2nd one defaults to SLCAN on Orange, but others default to mavlink.

russkel, 2023-11-21:
> On CubeOrange, serial0 and serial6. The first and last are USB. Check the hwdef.daf (see otg and otg2). 2nd one defaults to SLCAN on Orange, but others default to mavlink.

Sorry, what are you recommending we change the values to?

Ryanf55, 2023-11-21:
Sorry, for more context, that was notes from today's dev call.

 One reason I couldn't get DDS working over serial was because `SERIAL1_PROTOCOL` on CubeOrange/Orange+ is not the 2nd USB interface. 
See here for CubeOrange: https://github.com/ArduPilot/ardupilot/blob/69bfe9b8375feb94e657cf99ba3f2809dbfd9f8e/libraries/AP_HAL_ChibiOS/hwdef/CubeOrange/hwdef.inc#L30-L31

The defaults are MavLink on the second USB because it's configured so:
https://github.com/ArduPilot/ardupilot/blob/69bfe9b8375feb94e657cf99ba3f2809dbfd9f8e/libraries/AP_HAL_ChibiOS/hwdef/CubeOrange/defaults.parm#L10-L11

Cube Orange+ inherits the hwconfig from CubeOrange:
https://github.com/ArduPilot/ardupilot/blob/69bfe9b8375feb94e657cf99ba3f2809dbfd9f8e/libraries/AP_HAL_ChibiOS/hwdef/CubeOrangePlus/hwdef-bl.dat#L8
https://github.com/ArduPilot/ardupilot/blob/69bfe9b8375feb94e657cf99ba3f2809dbfd9f8e/libraries/AP_HAL_ChibiOS/hwdef/CubeOrangePlus/hwdef.dat#L9

BUT, it then sets SERIAL5_PROTOCOL to mavlink
https://github.com/ArduPilot/ardupilot/blob/69bfe9b8375feb94e657cf99ba3f2809dbfd9f8e/libraries/AP_HAL_ChibiOS/hwdef/CubeOrangePlus/defaults.parm#L10-L11

There is no USB serial available for DDS on CubeOrange+ as I was expecting. I was chasing the wrong issue. 

I need to make a cable to test your exact setup. Can you upload your params file and I'll copy wiring from that?

KyleJewiss, 2023-11-21:
Thank you for that. Here is the parameter file.
[parameters.zip](https://github.com/ArduPilot/ardupilot/files/13421906/parameters.zip)

tridge, 2024-01-03:
@KyleJewiss did you solve this? If not then please ping me on discord and we can look at it together

tridge, 2024-01-03:
> what is causing the issue Ryan is observing with the USB assigned to SERIAL1.

USB is never assigned to SERIAL1

russkel, 2025-03-22:
We unfortunately are no longer working on this project and not currently using Ardupilot. It sounds like @maripuusiim has the exact same issue @tridge. I hope he is able to assist you in solving this where we did not.

maripuusiim, 2025-03-24:
It is possible that this issue might not appear with GNSS modules that have a slower hot fix time. I’m open to performing further tests on my system if anyone has better suggestions than delay.

We have been using DDS over serial PPP network on a KakuteH7-Wing with Mosaic-H GNSS module, running the exact same firmware for several months. Initially, both GNSS antenna cables were 2.5 m long to achieve better moving base yaw performance. The DDS initialization issue appeared only after shortening the antenna cables to 0.5 m. Mosaic-H has a hot fix of ~20 s (factory specs).

Before adding an additional delay before main loop in `AP_DDS_Client.cpp`, we performed repeated tests in indoor-outdoor scenarios, as well as quick sequential startups in a van (without GNSS signal) versus under open sky conditions. In all cases, DDS initialization consistently succeeded only without GNSS signal. After adding 5s delay the initialization issue has not reappeared over several days and approximately 10 flights.

tizianofiorenzani, 2025-05-09:
I confirm that this issue is still open and unsolved.