ArduPilot/ardupilot #25548: Internal error 0x3000 after a video lost and RTL mode

juvinski, 2023-11-15:
## Bug report

**Issue details**
After loosing video signal, I put the plane in RTL mode and the Plane wasn´t ok - losing speed and altitude.

I have a plane with Ardupilot 4.4.1 with an O3 unit and I flew several - more than 30 times with this setup.
Today I was flying and I lost my video(O3+Goggles 2) - I activate the RTL and after a few moments the video came back and I perceived the plane was losing altitude and there was no power consuption - around 0.6 ma instead the 6 A, I retake the control using the FWBA mode and I couldn´t gain altitude - the elevons was ok. I land in a field and after 1 hour I could get the plane again, and arriving where the plane was I armed and the motor was spinning ok.
After download the logs I saw a messate Internal Error 0x3000

**Version**
4.4.1

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Flying wing - RVJet

**Hardware type**
_What Pixhawk 6C

**Logs**
https://drive.google.com/file/d/1rg9MHtfUNR4DX6zt5ken3fqewpM5lNqP/view?usp=sharing

rmackay9, 2023-11-15:
Thanks for the report.

Could you raise this in the [Plane support forum](https://discuss.ardupilot.org/c/arduplane/plane-44/171)?  I really don't want to be strict but otherwise our issues list turns into a support forum.

juvinski, 2023-11-16:
Hi @rmackay9 
I will, but I don't believe is a support issue, once the plane was ok for several flights and this was an unexpectect and really serious issue once the plane lost the motors after this error. But I will open in the forum.

best regards.

rmackay9, 2023-11-16:
@juvinski,

Thinking about it more, maybe you're right.  I don't think you should every see an internal error so I've re-opened this.

juvinski, 2023-11-16:
Hi @rmackay9 , thanks.
Digging a little bit more, appear that the IOMCU was reseted after an error 54 - and the field Nerr what means error in the mcu side is the 54.
[image]

I don't know if is related but was the first time I lost the video completely - I always when get a low tx rate or I increase altitude or even return, that was the first time, and at the O3 unit side I was getting a Air unit in low-power state - what just appear when the plane is disarmed -

[image]

Looking IOMCU Error X MCU Temperature X MCU Voltage, there is an oscillation in both - voltage and temperature at MCU side
[image]