ArduPilot/ardupilot #25625: More nmea data to transmit via COM port

AlksSAV, 2023-11-24:
## Feature request
To get more opportunities for boats, and the use of third-party electronic cards.  
At the moment MP allows you to output only GPS data to the COM port. It would be good to expand the capabilities by increasing the amount of information transmitted from all NMEA devices. This may be data from : AIS, echosounder, wind speed and direction, temprature.  At the moment I am making a device for[ measuring the speed of the boat](https://discuss.ardupilot.org/t/speed-through-water-stw-sensor-boat-speed-measurement/109413?u=mckey), the output of information will be according to the nmea protocol.

**Platform**
[ x] Rover
[ x] Boat
[ x] Sub

**Additional context**
This data can be used for electronic carts . [Usage example](https://discuss.ardupilot.org/t/navigation-charts-for-boat/81396?u=mckey).

magicrub, 2023-11-26:
Please be specific on what messages you're looking for

AlksSAV, 2023-11-26:
MP has such a menu, with GPS data output in NMEA format. 
[image]

Since Ardupilot also supports devices with NMEA, it would be good to also output this data to a single COM port. 
- AIS [AIVDM](https://arundaleais.github.io/docs/ais/nmearouter.html)
- Wind Speed and Angle [NMV](https://gpsd.gitlab.io/gpsd/NMEA.html#_mwv_wind_speed_and_angle)
- Echosounder [VHW](https://gpsd.gitlab.io/gpsd/NMEA.html#_vhw_water_speed_and_heading)

All this data is supported by Ardupilot. Mission planner should collect and output via one COM port, in addition to GPS data
Using the NMEA simulator, I previously[ created an example](https://discuss.ardupilot.org/t/navigation-charts-for-boat/81396?u=mckey).

AlksSAV, 2024-01-03:
A cheap  [speed sensor](https://github.com/AlksSAV/RC-Boat-Speed-Log/tree/main) has appeared.  - NMEA Water speed sensor.

AlksSAV, 2024-02-12:
Hi @magicrub! Are there any ideas, possibilities of implementation?