ArduPilot/ardupilot #25697: Cant build SITL on HW - possible result in moving of folder into Tools/scripts - .waf issue, then tmp issue

WilliamB2022, 2023-12-04:
## Bug report

**Issue details**
I am unable to run the SITL on HW as described in documentation. Appears to be an issue with the move of SITL on HW into Tools/scripts. Code within the sitl_on_hw.py file calls .waf from one level up, which is now 3 levels up. 

After amending this in the code fork, I am now faced with an issue with unable to find the .tmp folder. 
Rather than try hack away for myself, I thought id better raise it to be properly amended so that any future person looking to build SITL on HW may not come across the same issues.

khancyr, 2023-12-04:
hello,

What version are you using and what is exactly the output errors ? I have just tested the instructions and it is working fine 
https://ardupilot.org/dev/docs/sim-on-hardware.html#compiling-your-own-firmware-and-loading-to-autopilot

Arduhat, 2024-01-03:
Having the same issue. The instructions fail at [Compiling your own Firmware and Loading to AutoPilot](https://ardupilot.org/dev/docs/sim-on-hardware.html). It looks like it tries to look in/output to a temporary directory on line 34 of sitl-on-hw.py...

I have the dev environment correctly set up and a CubeOrange connected.

willbonholiday, 2024-01-04:
> hello,
> 
> What version are you using and what is exactly the output errors ? I have just tested the instructions and it is working fine https://ardupilot.org/dev/docs/sim-on-hardware.html#compiling-your-own-firmware-and-loading-to-autopilot

@khancyr I too am having the same issues. This is using a Cube Orange and even when trying to use the pre-compliled firmware that is detailed in the guide, the parameters given for the copter is way off and throws Accel and GPS calibration issues. 
The video attached to the guide is pretty good but just doesnt work that way when trying to build your own firmware. 

If anyone has a cube orange and can document / video how it is done using the precomplied firmware then please do share and enlighten us.