ArduPilot/ardupilot #25708: Copter: Waypoint "STOP & TURN" function

robustini, 2023-12-05:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

## Feature request

**Is your feature request related to a problem? Please describe.**
It would be very useful to implement a stop & turn function where the drone stops on the WP, turns in the direction of the next WP, and then continues in the direction of the next WP and so on.
This function is very useful for missions with magnetometer or LiDAR scanners where the drone is already facing the next waypoint at the beginning of the line.
This function is necessary to avoid losing data during scanning, as the drone now rotates as it travels to the next WP.
I remember many years ago this function was implemented, originally the Copter did this, and I was the one who asked to not stop over the wp to correct by yaw but to do it while moving.
Unfortunately, I now realize that this behavior could have been useful as well.
The video describes the functionality, achieved with the help of the condition yaw and delay events, which is quite tricky.

https://github.com/ArduPilot/ardupilot/assets/3864712/596c7c31-0e7d-4ed9-92be-4aa31d58f110

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

PUNEET-KOUNDAL, 2023-12-11:
Hey sir,

I'm looking to enhance waypoint navigation for a drone by implementing a stop-and-turn functionality. This would involve the drone pausing at a waypoint, turning to face the next one, and then resuming its mission. I've seen a video using "condition yaw" and "delay events" for this purpose.

Before I start the implementation, I have a few queries:

1. **Preferred Technologies:** Any recommendations on the technologies or libraries that work well for waypoint navigation and turning in a drone? I want to make sure the solution aligns with the existing tech stack.

2. **Precautions:** What precautions should I consider when implementing a stop-and-turn function? Are there potential issues, such as drift or calibration challenges, that I should be mindful of?

3. **Community Contributions:** Has anyone in the community worked on a similar feature? I'd like to leverage existing knowledge and possibly contribute to an ongoing effort.

4. **Alternative Approaches:** I'm leaning towards a stop-and-turn approach, but I'm open to other ideas. What's your opinion on using fan RPM for smoother turns? I'm concerned about the potential impact on balance.

I'd really appreciate your insights and any guidance you can provide!

peterbarker, 2025-05-11:
Sooo... this one can be done on a normal waypoint by setting param1 to a non-zero value.  It does mean a 1 second pause, but given the dynamics desired I don't think that's a big issue...

Closing this one - but please feel free to re-open if the 1-second thing is insufficient....

rmackay9, 2025-05-11:
Hi @peterbarker,

Thanks for all the efforts to clear up old issues. I think we should actually leave this one open because the request is that the vehicle proactively turn on the waypoint before proceeding.  At the moment, the vehicle will start turning **after** it starts towards the next waypoint.