ArduPilot/ardupilot #25715: Create ROS 2 interface for setting/getting geofence

Ryanf55, 2023-12-06:
## Feature request

**Is your feature request related to a problem? Please describe.**

The companion computer doesn't know what the geofence is on AP, and might try to command actions that violate the geofence.

**Describe the solution you'd like**
* Design a ROS interface for geofence management
* Follow REP-147

**Describe alternatives you've considered**

* Hard code geofence
* Use MavLink

**Platform**

- [x] All
- [ ] AntennaTracker
- [ ] Copter
- [ ] Plane
- [ ] Rover
- [ ] Submarine

haarshitgarg, 2024-03-05:
Hello @Ryanf55,

In order to achieve this can we subscribe and publish an appropriate message?

I was briefly looking at REP-147 and couldn't find a direct implementation that can be used. So, Do we need to create our own interface similar to those already there? I had a look at the AC_fence and there are a lot of parameters which could be  formulate the publishing message for geofence.

Ryanf55, 2024-03-05:
> Hello @Ryanf55,
>
> In order to achieve this can we subscribe and publish an appropriate message?
>
> I was briefly looking at REP-147 and couldn't find a direct implementation that can be used. So, Do we [...]

Yes, if you want to take a stab at writing a message definition that would be great.

timtuxworth, 2024-03-05:
Some thoughts to be included in a good definition of fences are included in this PR #25875  that @andyp1per and @Hwurzburg  have been working on regarding the current implementation.

The current implementation is not a good model, so it would be great if we can come up with something much better. Ideally for example, it should be possible to define behavior such as "return inside the fence" on breach - following a path that minimizes the amount of time spent outside the fence, and auto-reactivating the fence after the vehicle has returned inside the fence. (the fence is deactivated on breach ATM), as well as ideally avoiding the fence in the first place, which currently works for Copter (and Rover I think) but definitely doesn't work for Plane.