ArduPilot/ardupilot #25771: Plane (tailsitter): starts the descent on the home only after it reaches it

robustini, 2023-12-14:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

## Feature request

**Is your feature request related to a problem? Please describe.**
The most critical phase of a non-vectored tailsitter flight is landing.
After the transition in the case of (Q)RTL, the taisitter when passing into VTOL immediately begins the descent and repositioning to home.
These two simultaneous maneuvers often cause instability, especially in windy conditions.

**Describe the solution you'd like**
Instead, it would be very helpful if the tailsitter in the case of RTL behaved in this way:

1) Once the transition from plane to vtol is made, the code should store the height reached immediately after the transition angle
2) Reposition to the home point while maintaining the stored height
3) Start the descent for the land only after reaching the home

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

robustini, 2023-12-14:
Here you can clearly see the downhill curve after in QRTL, and in this case it was not very critical since evenc with crosswind came very close to the home point.

[image]

Here, however, the problem is clearly visible.

[image]

robustini, 2024-01-20:
This is my QRTL concept explained with a video, which is easier to understand and where the result is clearly visible, obviously simulated using manual transition and repositioning with sticks.
The tailsitter perform the transition in QRTL, as soon as it reaches the VT angle it stabilizes at that height for a few seconds, in this way the weathervaning feature has time to find if active the correct crosswind position, only then it is repositioned to the home keeping that initial height, once it reaches the point it starts the descent.
This is what it should do with the tailsitter.

https://github.com/ArduPilot/ardupilot/assets/3864712/a9bd517a-b1f4-4996-8940-5ad467f1b8d1

nishant-sg, 2024-02-10:
Hi, I was trying to reproduce the issue , which frame type did you use?

robustini, 2024-02-10:
> Hi, I was trying to reproduce the issue , which frame type did you use?

I use the eXplora frame created for my tailsitter but it is irrelevant, it does this with any non-vectored tailsitter.

nishant-sg, 2024-02-10:
Okay, I asked this becoz I am unable to get any quadplane to fly in SITL, as soon as I apply rc 3 input (throttle input) the map shows the plane spinning and on flightgear the plane is standing still. Infact the console HDG value keeps changing rapidly. Any idea why this happens ? Can you help?

robustini, 2024-02-10:
This is not the correct forum for this, you are better off asking on Discord in the channel that contemplates simulators.

nishant-sg, 2024-02-18:
Hi I have tried and implemented, the desired task, although I would like it to be verified before I submit a pull request.
What I have done is: once the tail sitter is in QRTL mode and it completes VTOL transition, then it will maintain its altitude until it has reached its correct landing spot( x,y coordinates) and then immediately start descent. I don't have any 3D sim, I have only tested it using SITL console and map.

I am sharing a video of what it looks like as of now.
In this example I have changed a lot of parameters to start the VTOL phase early and test the feature. As soon as you see position2 the vehicle is in VTOL mode and is slowly moving towards the landing spot maintaining the same altitude.

https://github.com/ArduPilot/ardupilot/assets/70840360/35c6c6cd-11e9-4cfd-9105-320f38024d1d

Let me know what you think of this.

robustini, 2024-02-19:
Nice result @nishant-sg!
From what I see on the sim it looks correct to me, the only thing is that when it reaches the home point it should not start the descent immediately but do so after a few seconds, not less than three, so that the tailsitter stabilizes before starting the descent.
If you have the code on one of your forks and share it I can test it with Realflight.

nishant-sg, 2024-02-19:
Hi! I have uploaded the changes to this fork (https://github.com/nishant-sg/ardupilot/tree/test). I have also reverted all the parameters back to normal. So I think it should be final product of what is required. Share the results, then if it requires a delay, will add that also.

robustini, 2024-02-20:
> Hi! I have uploaded the changes to this fork (https://github.com/nishant-sg/ardupilot/tree/test). I have also reverted all the parameters back to normal. So I think it should be final product of what is required. Share the results, then if it requires a delay, will add that also.

I saw your modification, thank you, and tested it.
It would never be approved since you do not create a condition for the tailistter but this behavior by modifying the quadplane.cpp would cover all VTOLs.
It does exactly what I asked for right after the transition, and in the video I o made in fact you see that when it transitions to VTOL it maintains the 18 meters, so correct behavior here.
From the code you have in fact removed the condition that contemplated repositioning in Z in case of QRTL, right.
But you also see that when it reaches home it immediately starts its descent and given the very high angle of 50 degrees it loses attitude.
In the second attempt I switch instead to QLoiter just before home and bring it down manually, that should do, i.e. arrived on home it should wait at least 3 seconds then start the descent, not immediately.
Also after transitioning to VTOL always at least 3 seconds pause and then start translating to home.

https://youtu.be/9-eGFlBw8I8

nishant-sg, 2024-02-21:
Hi,
I have gone through your concerns. I have added the tailsitter condition, so only tailsitters would be affected. Also I have added a 7second delay since 3 felt very less, but it can be easily changed in the code. Kindly test and please provide feedback.
https://github.com/nishant-sg/ardupilot/tree/tailsitter-qrtl-fix

robustini, 2024-02-22:
> Hi, I have gone through your concerns. I have added the tailsitter condition, so only tailsitters would be affected. Also I have added a 7second delay since 3 felt very less, but it can be easily [...]

Hi, I tested thelatest fix and always make a video to show its behavior, and I insert in the second RTL some wind to increase the difficulty.
As soon as the transition to VTOL is done it does not stop 5 seconds (I dropped from 7) as I had requested but proceeds towards immediately to home, it would be needed instead what I had requested.
Very good behavior instead when it gets to home, stops 5 seconds and performs LAND.
And you can see very well that now it doesn't go out of the flight envelope, it was as I thought, great!
So once the pause is inserted even immediately after the transition the change is perfect, it was as I predicted, well done!

https://youtu.be/HmkdrlRkClY

robustini, 2024-02-22:
Hi @nishant-sg, I tested the latest fix and always make a video to show its behavior, and I insert in the second RTL some wind to increase the difficulty.
As soon as the transition to VTOL is done it does not stop 5 seconds (I dropped from 7) as I had requested but proceeds towards immediately to home, it would be needed instead what I had requested.
Very good behavior instead when it gets to home, stops 5 seconds and performs LAND.
And you can see very well that now it doesn't go out of the flight envelope, it was as I thought, great!
So once the pause is inserted even immediately after the transition the change is perfect, it was as I predicted, well done!

https://youtu.be/HmkdrlRkClY

robustini, 2024-02-23:
> So just to clear up before next changes, you want a 5 second delay after transition to VTOL and then a 5 second delay when it reaches the home location, correct?

Correct!

robustini, 2024-02-25:
@nishant-sg is it complicated to put 5 seconds of delay even after the transition to VTOL?
Analysing your modification above, I think that it is not possible to insert it easily in quadplane.cpp, and that perhaps it is better to act in the tailsitter.cpp in the section concerning the VTOL transition, putting the delay after the VTOL angle is reached but obviously with the condition "if qrtl".

robustini, 2024-02-25:
@nishant-sg complex thing to insert delay also after transition to VTOL?

robustini, 2024-02-27:
@nishant-sg I did some tests, I think the point where the pause should be inserted is obviously in 'case QPOS_POSITION1:', but obviously with the tailsitter condition in QRTL.

nishant-sg, 2024-02-27:
Hi, sorry for the delay got stuck somewhere. It might not be that tough to code, what I find more challenging is testing tail sitter since it is very hard to takeoff , so it creates a huge delay in the coding process as well. I will try and do something today.

robustini, 2024-03-01:
@nishant-sg the log and the video of your latest fix.
In the second attemp, I increase one transition parameter to make it happen sooner.

[00000047.ZIP](https://github.com/ArduPilot/ardupilot/files/14465371/00000047.ZIP)

https://youtu.be/uJIEtrpfwJY

robustini, 2024-03-02:
@nishant-sg I have yet to investigate but I have seen a problem that now occurs if the tailsitter passes the home point during QRTL.
Instead of staying at the preset height it starts to climb many metres, and then regularly descends.
In the video I show what happens by varying the Q_TRANS_DECEL parameter, if set to 6 (the default, wrong for that tailsitter) it overtakes the home point a lot but does not generate the problem, the same thing if set to 1, i.e. it anticipates the VT transition a lot.
But if I set it to 4, i.e. it makes the transition practically on the home point, the problem is generated.
Here the video and the log.

https://youtu.be/CDkLzWjN6Zs

[00000004.ZIP](https://github.com/ArduPilot/ardupilot/files/14468346/00000004.ZIP)

robustini, 2024-03-03:
@nishant-sg this value was fine before, at 10, and that climb does not happen.

// distance that we switch to QPOS_POSITION2
    const float position2_dist_threshold = 10.0

robustini, 2024-03-04:
@nishant-sg I realised why in some conditions the tailsitter was rising in the 7-second pause.
I solved it by forcing the climb to 0 there as well.
So in the quadplane.cpp it will be:

case QPOS_LAND_FINAL: {
        if (tailsitter.enabled() && now_ms-last_pos2_ms<7000) {
			   set_climb_rate_cms(0);  
               break;

nishant-sg, 2024-03-04:
So is the issue completely resolved? are you able to stabilize just after coming out of forward flight?
If so then should we close this issue?

robustini, 2024-03-04:
> So is the issue completely resolved? are you able to stabilize just after coming out of forward flight? If so then should we close this issue?

Yes thanks, do you do a PR on this?

robustini, 2024-03-04:
> Thanks to you for your support and the especially providing such wonderful feedbacks. Also if possible, please share video of the final changes. Yes I will do a PR on this now.

I await your PR and comment with a video, without this modification and with.

IamPete1, 2025-05-06:
As https://github.com/ArduPilot/ardupilot/pull/26417 has not been merged I'm re-opening.

closed 2026-09-07 by tridge

IamPete1, 2026-09-07:
We now have `Q_RTL_PAUSE_TIME` for this.