ArduPilot/ardupilot #25779: Cannot compile ArduPilot. PermissionError: [Errno 13] Permission denied: 'tasklist.json'

buchbergerd, 2023-12-15:
## Bug report

**Issue details**

I would like to cross-compile the ardupilot firmware on Ubuntu 22.04 to include OpenDroneID, but waf raises an Exception.
Following [https://ardupilot.org/dev/docs/building-setup-linux.html](https://ardupilot.org/dev/docs/building-setup-linux.html) I set up my the build environment:

```sh
cd  ~
git clone --recurse-submodules https://github.com/ArduPilot/ardupilot
cd ardupilot
docker build . -t ardupilot
docker run --rm -it -v `pwd`:/ardupilot ardupilot:latest bash
```

After that, I wanted to start building following [https://github.com/ArduPilot/ardupilot/blob/master/BUILD.md](https://github.com/ArduPilot/ardupilot/blob/master/BUILD.md):

```sh
./waf configure --board CubeOrange
```

but it just throws an Exception:

```
ardupilot@d06946f6c286:/ardupilot$ ./waf configure --board CubeOrange
Traceback (most recent call last):
  File "/ardupilot/modules/waf/waflib/Scripting.py", line 158, in waf_entry_point
    run_commands()
  File "/ardupilot/modules/waf/waflib/Scripting.py", line 248, in run_commands
    run_command('init')
  File "/ardupilot/modules/waf/waflib/Scripting.py", line 235, in run_command
    ctx.execute()
  File "/ardupilot/modules/waf/waflib/Context.py", line 204, in execute
    self.recurse([os.path.dirname(g_module.root_path)])
  File "/ardupilot/modules/waf/waflib/Context.py", line 286, in recurse
    user_function(self)
  File "/ardupilot/wscript", line 92, in init
    generate_tasklist(ctx, False)
  File "/ardupilot/wscript", line 634, in generate_tasklist
    with open(os.path.join(Context.top_dir, "tasklist.json"), "w") as tlist:
PermissionError: [Errno 13] Permission denied: 'tasklist.json'
```

**Version**

Ardupilot: `Cloned the latest version of the git repo.`
Docker: `Docker version 24.0.7, build afdd53b`
Ubuntu: `Linux <hostname> 6.2.0-39-generic #40~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov 16 10:53:04 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`

**Platform**
[X] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
n/a

**Hardware type**
Cube Orange

**Logs**
n/a

Thanks in advance for help in fixing this error / the documentation.

khancyr, 2023-12-18:
working fine for me with those step.
are you using root user in your host ?
You can try `git reset --hard HEAD` and `git submodule update --init --recursive -f ` to reset any changed you had made and try again.

buchbergerd, 2023-12-19:
Thanks for trying! Unfortunately, resetting any change did not alter the error message.

I use a normal user (but it is allowed to use sudo, as well as use docker without typing sudo).
The user's groups: `users dialout sudo netdev docker`

I searched for the file `tasklist.json`, but I did not find it. Where is it supposed to be?
It is neither in my cloned folder nor in the waf github repository [https://github.com/ArduPilot/waf/tree/1b1625b8e7da6e1307d73335cb995fa8813d5950](https://github.com/ArduPilot/waf/tree/1b1625b8e7da6e1307d73335cb995fa8813d5950) (used the "Go to file" search bar).

The output of the reset:
```bash
X@Y:~/ardupilot$ git reset --hard HEAD
HEAD is now at 61b3ad326d AP_Scripting: added SCR_THD_PRIORITY parameter
X@Y:~/ardupilot$ git submodule update --init --recursive -f
Submodule path 'modules/ChibiOS': checked out 'd8c45abde5fb398ef9da6fc8fc36992f2ae7b15c'
Submodule path 'modules/ChibiOS/ext/lwip': checked out '0a0452b2c39bdd91e252aef045c115f88f6ca773'
Submodule path 'modules/CrashDebug': checked out '599965086437137ec0fe66e185611f43f335f889'
Submodule path 'modules/CrashDebug/CrashCatcher': checked out '4cf6e11df3a081b0f573a7834ed8e1e4000af73e'
Submodule path 'modules/CrashDebug/CrashCatcher/CppUTest': checked out '011f371fdb5d7e441fb023be67295783e053df4a'
Submodule path 'modules/CrashDebug/mri': checked out 'b35c89451b41a45b7f5b580acaff555315ac88e2'
Submodule path 'modules/CrashDebug/mri/CppUTest': checked out '011f371fdb5d7e441fb023be67295783e053df4a'
Submodule path 'modules/DroneCAN/DSDL': checked out 'de93d9c8bb76de1b093050e3a31e4d4ce539c577'
Submodule path 'modules/DroneCAN/dronecan_dsdlc': checked out 'ebaf96860a11a4cc43c01df6b651df143c6cde2d'
Submodule path 'modules/DroneCAN/libcanard': checked out '22102c717db29cc2a2c2869ff80f3e4389704d89'
Submodule path 'modules/DroneCAN/libcanard/drivers/avr/avr-can-lib': checked out '9c6bc9118de66d6edaf1b8539e2b9717ba26d123'
Submodule path 'modules/DroneCAN/pydronecan': checked out '19fdf2e5b383243ccdb1094edae0603cf11469e8'
Submodule path 'modules/Micro-CDR': checked out '3d1b17703c7cf4f22def2910bc845bdb5152d7b5'
Submodule path 'modules/Micro-XRCE-DDS-Client': checked out '97175304425c5bee87c6fddd99de1ef8d0c394dc'
Submodule path 'modules/gbenchmark': checked out 'd572f4777349d43653b21d6c2fc63020ab326db2'
Submodule path 'modules/gsoap': checked out 'e1f690585d4803402584962bfaa8240ecaf1db30'
Submodule path 'modules/gtest': checked out 'c5fed93f941865a0e912e9baf46ded713506590a'
Submodule path 'modules/mavlink': checked out '71af5c43fc24ffb237d3a93dc0c8358d67aa0cc0'
Submodule path 'modules/mavlink/pymavlink': checked out '6b468fb13abb7c5bdb6b7dfb7cbac5af2e0a37ed'
Submodule path 'modules/waf': checked out '1b1625b8e7da6e1307d73335cb995fa8813d5950'
```

paulerikf, 2024-01-29:
I'm running into the same issue - somewhat at a loss for what to do. It seems related to #20850 maybe?