ArduPilot/ardupilot #25781: Feature request for mavros full support

ZhongmouLi, 2023-12-16:
## Feature request
Add more mavros support and make more mavlink messages published in mavros

**Is your feature request related to a problem? Please describe.**
mavros wrappers mavlink messages taken at ArduCopter and make them visible in ros. However, there are only few messages are supported by ArduCopter, like imu data, imu data raw, rc_in even their frequencies are limited to 50Hz. It makes development very hard when additional information are needed, like attitude_target information that shows what throttles and attitudes the controller is trying to track. This information is so import to tune controllers in ros. 

This could be very attractive to UAV researchers using ros. 

**Describe the solution you'd like**
Please make ArduCopter send ATTITUDE_TARGET messages and make it publish to mavros/setpoint_raw/attitude_target.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

IamPete1, 2023-12-16:
Is there anything we need to do on ArduPilot side? I think this just just a macros feature request? Maybe open a issue over there? https://github.com/mavlink/mavros

Our ROS2 support is advancing very fast, https://youtu.be/SISkW_7g57M?si=Vn0g5thL49JVG_a1

ZhongmouLi, 2023-12-17:
Hi, @IamPete1 thanks for your reply.

I just checked [GCS_MAVLINK](https://github.com/ArduPilot/ardupilot/blob/Copter-4.4/ArduCopter/GCS_Mavlink.cpp) that there is a function called ```GCS_MAVLINK_Copter::send_attitude_target()``` which seems to send ATTITUDE_TARGET to mavlink. 

I guess GCS_Mavlink.cpp of ArduCopter manages all information sent to mavlink. Do I understand correct?

peterbarker, 2025-05-11:
> I guess GCS_Mavlink.cpp of ArduCopter manages all information sent to mavlink. Do I understand correct?

Not really.  It handles a lot of it - but the base library in `libraries/GCS_MAVLink` does most of it now-adays.

Several of our libraries also send their own mavlink messages, too.

Did you still need something done on the AP side here?