ArduPilot/ardupilot #25784: AP_DDS UDP on ChibiOS doesn't work yet

Ryanf55, 2023-12-17:
## Bug report

**Issue details**

I spent some time working with @andrewtridgell  today; DDS over UDP on CHibiOS is still not working. 

Mavproxy logs:
```
AP: DDS Client: Network up, resuming main_loop
AP: DDS UDP INIT ../../libraries/AP_DDS/AP_DDS_UDP.cpp:97
AP: Connecting to 192.168.1.5:2019
AP: ../../libraries/AP_DDS/AP_DDS_Client.cpp:602
AP: DDS: Init complete
AP: Failed to read 512 ret=-1 errno =11, timeout=200
AP: DDS: Topic/Pub/Writer session request failure for index '0'
AP: DDS: Status '0' result '255'
AP: DDS: Status '1' result '255'
AP: DDS: Status '2' result '255'
AP: DDS Client: Creation Requests failed
```

I've attached logs from SITL and 6X to compare from the microROS agent in debug mode. You can see that that the exchange is basically the same, but then the 6X stops responding. 

**Version**
Branch: WIP/dds-fix-chibios-udp

**Platform**
[*] All
[ ] AntennaTracker
[ ] Copter
[ ] Plane
[ ] Rover
[ ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
Pixhawk 6X

**Attempted resolutions**
* Bump up timeouts

[udp_6x.log](https://github.com/ArduPilot/ardupilot/files/13694295/udp_6x.log)
[udp_sitl.log](https://github.com/ArduPilot/ardupilot/files/13694299/udp_sitl.log)

tridge, 2024-01-04:
It's working fine for me with todays master:
[image]

Ryanf55, 2024-02-20:
> It's working fine for me with todays master: [...]

Which hardware are you on? I can't believe I missed this.

Ryanf55, 2024-03-17:
I tested on Pixhawk6X, works as advertised. It must have been an earlier bug in the network code. 
Because it's now working, I've created a tutorial:
https://github.com/ArduPilot/ardupilot_wiki/pull/5828

The following steps are what I tested today on master hash 40dd8139eb7ffecd83a2c5cc8dd5ed1c4cd928ab.

```
./waf configure --board Pixhawk6X --enable-dds
./waf plane --upload
mavproxy.py --console
param set DDS_ENABLE 1
# Check the hwdef file for which port is OTG2 for SERIAL Comms
param set SERIAL8_PROTOCOL 45
param set SERIAL8_BAUD 115
# OR, use Ethernet
param set NET_IPADDR0 192
param set NET_IPADDR1 168
param set NET_IPADDR2 1
param set NET_IPADDR3 6
param set DDS_IP0 192
param set DDS_IP1 168
param set DDS_IP2 1
param set DDS_IP3 5
reboot
```

closed 2024-03-17 by Ryanf55