ArduPilot/ardupilot #25792: EAHRS is performing duplicate temperature compenstation

Ryanf55, 2023-12-18:
## Bug report

**Issue details**

Certain devices exposed through EAHRS, like the MicroStrain7, supply data that is already temperature compenstated.

https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/external_content/dcp/Data/sensor_data/data/mip_field_sensor_scaled_mag.htm
https://s3.amazonaws.com/files.microstrain.com/GQ7+User+Manual/external_content/dcp/Data/sensor_data/data/mip_field_sensor_scaled_accel.htm

When the External AHRS driver posts data to sensor libraries, they perform temperature compenstation on it again, as long as `HAL_INS_TEMPERATURE_CAL_ENABLE` is defined, which is based on board size.  

_Please describe the problem_

This causes double temperature compensation, leading to incorrect values.

**Proposed Solution**
Add `bool is_temperature_compensated;` to `AP_ExternalARHS::ins_data_message_t`, and then add the necessary hooks in `AP_InertialSensor::handle_external(const AP_ExternalAHRS::ins_data_message_t &pkt)` to stop doing temperature compensation if the supplied value is already temp-compensated.

**Version**
master `1e644d56a51806b63dcd68d5e6566ec0f10248fc`

**Platform**
[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

Ryanf55, 2023-12-19:
instead of a bool, use a 3-value

```
IsTempCalibrated
IsNotTempCalibrated
DoesntProvideTemp
```

Airspeed already handles this well, can copy the logic.

peterbarker, 2025-05-11:
@Ryanf55 going to circle back to this one?  Looks like it got a positive reception...

Ryanf55, 2025-05-11:
I don't have time at this point. Anyone is welcome to submit changes to support this. Given it's a small change, I think this would be a good first issue.

ohyaiamhere, 2025-08-25:
> I don't have time at this point. Anyone is welcome to submit changes to support this. Given it's a small change, I think this would be a good first issue.

I have submitted the changes mentioned above, in a new pull request.