ArduPilot/ardupilot #25809: Pixhawk 6c FMU B/E Led Assignment and Rover 4.4 stable

nvarona-ied, 2023-12-20:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

_Please describe the problem_

After installing the stable Ardurover 4.4 firmware, I have the B/E FMU red LED blinking, is there any way to remove the red LED blinking? or know what happens to him. Someone commented in another ISSUU that the color of the LED was assigned incorrectly and the blue one would have to be turned on, but that it was a code issue. Everything is correct ?

**Version**
_What version was the issue encountered with_

ArduRover 4.4 stable

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
PIXHAWK 6C

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2023-12-22:
I think that in some cases the blinking LED may go away once the calibrations are completed (RC calibration, etc).

nvarona, 2023-12-22:
Thanks, I'll look into it, what if I don't have any device plugged into the RC channel to be able to calibrate?

rmackay9, 2024-01-09:
@nvarona,

In general we think it's always a good idea to have an RC transmitter.  A joysticks still counts as an RC transmitter though.

rmackay9, 2024-01-09:
@nvarona,

If you change some of the RC1_MIN, MAX values then you can probably get the LED to be normal.

The problem with not having an RC transmitter and also no joystick is really safety.  Especially for new builds where you've never flown the vehicle yourself, it's really quite dangerous to have no way for a user to quickly retake manual control.

nvarona, 2024-02-06:
In the end I was able to buy a FRSKY XMPlus module and I have it "BIND" with my Taranis flashing the B/E LED of the FMU part. I have also tried it on a pixhawk 6mini and it does the same with the different versions of Ardurover 4.4, although I have version 4.4 Stable on.

I had read that it was possible a development error of turning on the blue LED, and now the red one turns on.

I have noticed that the pixhawk 6c turns on without that red LED and at the end of booting, after the Buzzer sounds, it flashes.

Any more ideas?
[image]

soul-blues, 2024-10-02:
Hello everyone. Pixhawk 6c: Yesterday everything works, all LEDs are illuminated properly and servos ran properly; switched off. The next day everything is switched on and no servo movements etc. But the FMU LED flashes permanently red. Does anyone know anything that can help? Greetings Soul.Blues