ArduPilot/ardupilot #25825: Rover: Circle mode instabilities

stephendade, 2023-12-23:
Whilst attempting to tune my rover, I experienced some serious instabilities and "wandering off" in circle mode:

5m circle at 2m/s speed:
[image]

A 5m circle at 1m/s speed was fine.

My current theory is that the default steering PID values are too low for this particular vehicle. I've confirmed in manual mode that the vehicle can definitely turn at the rates needed for a 5m circle at 2m/s (23deg/sec).

I could replicate it in SITL via restricting the ``ATC_STR_ACC_MAX`` and ``ATC_STR_RAT_MAX`` parameters to below the minimum degrees/sec required for the circle. So:
```
param set CIRC_RADIUS 5
param set CIRC_SPEED 4
param set ATC_STR_ACC_MAX 90
param set ATC_STR_RAT_MAX 30
```
SITL result:
[image]

I usually wouldn't be concerned about an untuned vehicle performing badly in these higher level flight modes, but given the Quiktune feature requires running in circle mode, it should be expected that untuned vehicles will be running in this mode.

For a fix, I'd suggest that circle mode needs to stop the vehicle if it gets more than X metres outside the circle (where X is +20% on radius?) and suggest to the user that they either a) Increase the radius b) decrease the speed c) increase the Steering PID FF(?).

Alternatively, maybe the way circle mode calculates it's trajectory needs to be improved? I feel that it does wander more than necessary ... it's like it gives up after a while!

rmackay9, 2023-12-26:
yes, I've seen these problems as well actually but haven't found a way to protect against the problem.  I think the issue occurs because we pass a target position, velocity and acceleration to the position controller but even a large position error does not overpower the velocity and acceleration inputs.  So maybe the solution should be that the desired velocity and acceleration only get to use whatever control bandwidth is left after the velocity and acceleration required to shrink the position error has been calculated.

stephendade, 2024-01-02:
As an update to anyone with a similar problem: Increasing ``ATC_STR_RAT_FF`` can "fix" the issue until a quiktune can be run.

closed 2024-09-17 by stephendade