ArduPilot/ardupilot #25833: LOITER_UNLIM and LOITER_TIME not working on arducopter

HunterSpy, 2023-12-26:
## Bug report

I am currently engaged in the development of MAVROS and SITL. While working on configuring a 'loiter unlimited' and 'loiter times', I have encountered difficulties with the associated command codes. Interestingly, the issue does not persist when implementing the "loiter turns" command, as the drone successfully executes the desired behavior.

Specifically, when attempting to employ the "loiter unlim" or "loiter turns" commands, the drone successfully takes off and proceeds towards the designated waypoint. However, it fails to initiate the expected circling behavior at the specified waypoint as intended in the mission plan.

I am actively seeking assistance or insights to address this anomaly, particularly within the context of ArduCopter. Your guidance on rectifying this issue would be highly valuable and appreciated.

HunterSpy, 2023-12-26:
[image]

I am using the above documentation to edit the parameters.

rmackay9, 2023-12-26:
Hi, so this is just a terminology issue I think.  For Copter Loiter doesn't mean circle because Copters don't need to circle to maintain a position.

closed 2023-12-26 by rmackay9

HunterSpy, 2023-12-26:
Thanks for the reply, then can you suggest or guide me how to perform this mission, that my copter will circle around at some specific waypoint, as the picture shown below ? 

[image]

rmackay9, 2023-12-26:
Hi @HunterSpy,

LOITER_TURNS should work for this.

HunterSpy, 2023-12-29:
Thanks again, 
but in LOITER_TURNS the drone will follow a circular trajectory, for the number of turns which is predetermined in the configuration, rather than maintaining continuous circular motion

rmackay9, 2023-12-29:
Hi @HunterSpy,

yes, that's right.  unfortunately our mission library is not able to store more than about 14bytes per command which means we cannot store all the possible settings that you might which is why we need to fall back to use the parameters.  This is a long standing issue that is somewhere on [the issues list](https://github.com/ArduPilot/ardupilot/issues) but I can't immediately find it.