ArduPilot/ardupilot #25846: There is a bug in AP_AHRS_DCM::drift_correction()

luweiagi, 2023-12-29:
from here 
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_AHRS/AP_AHRS_DCM.cpp#L1024

```c++
if (now - _last_wind_time > 2000 && airspeed_sensor_enabled()) {
    // when flying straight use airspeed to get wind estimate if available
    const Vector3f airspeed = _dcm_matrix.colx() * AP::airspeed()->get_airspeed();
    const Vector3f wind = velocity - (airspeed * get_EAS2TAS());
    _wind = _wind * 0.92f + wind * 0.08f;
}
```

we can see that `AP::airspeed()->get_airspeed()` means EAS NOT TAS, because `TAS = EAS * get_EAS2TAS()`.
but,  at here
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_AHRS/AP_AHRS_DCM.cpp#L710

```c++
        float airspeed = _last_airspeed;
#if AP_AIRSPEED_ENABLED
        if (airspeed_sensor_enabled()) {
            airspeed = AP::airspeed()->get_airspeed();
        }
#endif

        // use airspeed to estimate our ground velocity in
        // earth frame by subtracting the wind
        velocity = _dcm_matrix.colx() * airspeed;

        // add in wind estimate
        velocity += _wind;
```

we can see that `AP::airspeed()->get_airspeed()` is treated as TAS, because it is added with wind directly.
So the above code snippet can be rewritten as:

```c++
        float airspeed = _last_airspeed;
#if AP_AIRSPEED_ENABLED
        if (airspeed_sensor_enabled()) {
            airspeed = AP::airspeed()->get_airspeed();
        }
#endif

        // use airspeed to estimate our ground velocity in
        // earth frame by subtracting the wind
        velocity = _dcm_matrix.colx() * airspeed  * get_EAS2TAS();

        // add in wind estimate
        velocity += _wind;
```

So, the question is AP::airspeed()->get_airspeed() means TAS or EAS on earth?

luweiagi, 2023-12-29:
@peterbarker I don't know who is responsible for maintaining and improving the code of EKF and DCM

amilcarlucas, 2024-01-04:
that would probably be @priseborough

luweiagi, 2024-01-08:
Hi,  Paul@priseborough , does this bug exist?

peterbarker, 2024-11-24:
@luweiagi thankfully I don't think that second bug actually exists.  `new_value` is only set `true` after we have assigned into `yaw_deltat`.

The structure here is... sub-optimal.  But no bug, I think.

peterbarker, 2024-12-21:
@luweiagi we've merged the fixes for this problem now - many thanks for the report!

closed 2024-12-21 by peterbarker