ArduPilot/ardupilot #25853: Request: QSPI flash storage

David-OConnor, 2023-12-30:
I am requesting support for QuadSPI (QSPI) flash support on ArduPilot for use with Flash storage, as an alternative to SPI flash and SD Cards. QSPI is a protocol explicitly designed for use with flash memory.

Reasons:
- QSPI is a faster alternative to SPI flash, and can use memory mapping
- QSPI has no downsides compared to SPI
- All STM32 MCUs supported by AP have a QSPI peripheral
- Most (all?) memory chips supported by AP are capable of QSPI.

As an engineer, I would, choose QSPI over SPI for flash storage for any project that uses an MCU with QSPI support. So, it was surprising that AP supports traditional SPI for flash, but not QSPI.

[STM32H743 Reference manual](https://www.st.com/resource/en/reference_manual/dm00314099-stm32h742-stm32h743-753-and-stm32h750-value-line-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf) See Chapter 23: QUAD-SPI Interface

magicrub, 2023-12-31:
Why do you think it's not supported? Have you checked? I count 7 hwdef targets in master that use it. Search for a hwdef that contains `QUADSPI_CLK`.

closed 2023-12-31 by magicrub

David-OConnor, 2023-12-31:
Could you please confirm that it is supported for data storage? My understanding is this is not the case, and it is only supported for program storage. You'll note that the targets you list are for ones with the <2Mb flash AP requires.

magicrub, 2023-12-31:
Well, you're half right. None of them use them as a dedicated non-volatile memory for params and missions. We've gone to an even faster model: use the internal program flash of the STM. A nice example of this is CubeRed where it uses 2x 128kB flash pages in the H757 for storage but then also has [QSPI for another lots-of-megabyte storage](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/CubeRedPrimary/hwdef.dat#L191C31-L191C31) for program memory.

There's actually no advantage of using QSPI for the small amount of flash we use for params and missions. For logs, that will never be enough, you'd need eMMC or an SD card for multi-gigabyte storage.. so QSPI eeprom ends up just being a waste of pins.

What did you have in mind? [Join us on our weekly developer call](https://ardupilot.org/dev/docs/ardupilot-discord-server.html#weekly-dev-calls) on Discord to talk about it!