ArduPilot/ardupilot #25871: Mac Environment Build Instructions do not work for Sonoma

arshish1612, 2024-01-02:
## Bug report

**Issue details**
1. The existing instructions to setup the build environment for MaxOX 14 do not work. Issues are faced with incorrect Python Version, incorrect G++ Version. 
Link & Video Referred to: https://ardupilot.org/dev/docs/building-setup-mac.html

2. The Automatic environment script for mac does not work either. I dont have the exact errors just now. I will have to re-run it on another mac to get the errors again.

Please note that in both cases I was attempting to setup the environment from scratch. 

**Version**
N/A - ./waf configure fails

**Platform**
N/A - ./waf configure fails

**Airframe type**
Copter however the issue would arise for any frame

**Hardware type**
Pixhawk6X however this is irrelevant becase - ./waf configure fails

**Logs**
I dont have the logs at the moment as the issue is now fixed but the problem is related to brew installing an older version 5.4.1 of G++ and the multiple conflicting versions of python. 
a. .pyenv did not work
b. after removing .pyenv version 3.9.6 did not work
c. finally pointed to 3.11.6 under homebrew and that worked.

srmainwaring, 2024-01-17:
@arshish1612, thanks for the report. These instructions are very out of date and need a complete overhaul. I'm rebuilding an old intel MacBook from a clean install of Big Sur, so I'll document the install for that as a baseline and update for Sonoma Intel and M1 after.

I don't use pyenv as I have found issues running ROS 2 with the shims used by pyenv. On the other hand a standard virtual environment created using `python3 -m venv` works well for ArduPilot and ROS, and I'd recommend that. It's important to start with clean brew `site-package` folders though, which may involve uninstalling python, clearing the existing packages, and reinstalling.

ShadowWalker8642, 2024-03-11:
i ran the prereq script works pretty well install python using pyenv

peterbarker, 2025-05-11:
OK, so @ShadowWalker8642 says that the environment install script is OK.

@srmainwaring - should we just remove the instructions in the Wiki and replace with "either run the environment install script or do what it does"?

srmainwaring, 2025-05-11:
> [@srmainwaring](https://github.com/srmainwaring) - should we just remove the instructions in the Wiki and replace with "either run the environment install script or do what it does"?

Sounds good to me. Has the advantage that the instructions are effectively in one place, and a script is more likely to be kept current.