ArduPilot/ardupilot #25895: Various Issues with my SITL

devnim, 2024-01-04:
-Window 11, using WSL1. (I had to downgrade earlier from WSL2 because of another issue - #25257 where I added comments.
- Ubuntu 20.04.6 LTS. (had Ubuntu 22 LTS version and changed to 20.04 version to try to fix the issues)
- I have a new desktop with NVIDIA 3060 GPU. 

I am opening this issue after I posted on the forum: https://discuss.ardupilot.org/t/installed-from-main-git-but-got-4-5-0-dev-also-an-issue-on-copter-drifting/111196/3

I continue to experience various little issues with SITL. I experienced it on non-graphical SITL, and also when switched to Airsim and Webots. One issue is that the copter in SITL continues to drift after I send "attitude 0.9961947 0.0871557 0 0 0.5" command. One time it was drifting at a high speed and for a long time and then crashed. This was on ArduCopter 4.5.0-Dev. Then I followed the instructions given in the aforementioned forum discussion. I think it may have slowed down the drift. Sometimes it stops and sometimes it slowly keeps drifting. As seen in the log file. Another issue is with when I enter Circle mode. One time when I reached an altitude of e.g. 10 meters in guided mode and switched to circle mode, it circled while losing altitude and crashed. Another time, it worked fine. Today, it crashed as soon as I entered circle mode from the guided mode. 

I reported an earlier issue where the copter was yawning right after takeoff (https://discuss.ardupilot.org/t/copter-in-sitl-is-spinning-yawing-after-takeoff/111173/2). I reinstalled Ubuntu, rebuilt Ardupilot, etc. It solved the yawning issue, but I noticed the drift, circle mode issue. 

Another issue I noticed is that Maxproxy command line freezes up sometimes and does not respond to keyboard input and I have to restart it. Also I have experienced various pre-arm failsafe on SITL, such as compass not calibrated, something with GPS, something with rate not matching (222 < 300hz ? ). I have had to disable these check to fly. Sometimes I do not get these messages, such as in the log below. There were no pre-arm failsafe. 

I created a folder and added a link. It contains the most recent log and should contain pre-arm logs. It show the circle mode issue and the drifting. In the first attitude command, the drift stops eventually but in the second, it keeps going. The folder also had output from the WSL window and a Windows Console. I first connected Mavproxy. I added note in the output file to note where I connected and when disconnected. I connected Mission Planner afterwards.  

https://drive.google.com/drive/folders/16CWAyuCly8AL4i6EuMQiV3SS7XQPkNBZ?usp=sharing

This folder contains earlier logs. I am not sure if that was when I had ArduCopter 4.5.0-dev - 
https://drive.google.com/drive/folders/1Z7PGrUFB_d3fedcX9JDeDg7U9iRY1uwE?usp=sharing

khancyr, 2024-01-04:
Hello,

You should try with normal SITL that your setup is working fine before trying to use another simulator.

Drone crashing in auto control mode is generally a performance issue : lacks of ram or AV software nuking the performance, so try to look after those.

About mavproxy, how do you connect it ? From the log it looks like you never disconnected it, so you are using another terminal to connect another mavproxy instance ? Where do you also connect mission planner?

devnim, 2024-01-04:
Thanks for the response. I am having this issue on normal SITL, without connecting webots or airsim.

I doubt that my ram is an issue. I have 32GB ram and I only have web browser open when I run normal sitl. 

I unintstalled McAfee prior to all of this. I only have standard AV that comes with windows. However, I sometimes feel my web browser anti-virus is a bit aggressive in blocking website. I will check again. (Btw, as I mentioned when I had issue earlier with WSL2, I also thought it could be firewall/av and I disabled firewall to try and it didn't solve the prior issue.) 

My usual workflow has been first, to run sim_vehicle command on WSL. However, the --console and --map flag do not produce these windows. Not sure if this is normal for WSL1? So I needed somerhing like Mission Planner for seeing the map, etc. Also, the wsl console where I ran sim vehicle, thebstreaming messages, e.g. heigh ..., would cut off my typed commands. So that is why the recent use of an external maxproxy. Also is Mission Planner considered another instance of mavproxy? 

I have not seen not anywhere in the Ardupilot's instructions that I should disconnect the mavproxy in my one WSL console before connecting to Mission Planner.  I will see if I can disconnect from the first mavproxy before connecting to MP (and not another mavproxy instance). I will let you know if it solves the issues. Thanks.

> Hello,
>
> You should try with normal SITL that your setup is working fine before trying to use another simulator.
>
> Drone crashing in auto control mode is generally a performance issue : lacks of ram [...]

devnim, 2024-01-05:
Update: I tried running normal SITl and not connect to either another Mavproxy or Mission Planner. When I reached the altitude 10 in guided mode, I switched to Circle mode and the copter fell to the ground. 

I then closed the WSL, made sure it was not running in the background by doing wsl --shutdown. I opened Mission Planner (Windows version - 1.3.81.....Arducopter V4.4.4). I activated the guided mode, armed without any issues, took off to 10 meters, switched to Circle mode and ... it crashed to the ground. I save the log file of this flight here: https://drive.google.com/file/d/1A2HZglC1JYdcGLecA9OaLzAJj-j5PH_W/view?usp=drive_link

Also I am including a snippet of messages from Mission Planner's console (the black console that opens up with mission planner) here, from the time I set mode to Circle to the time I crashed and then disarmed. 

INFO MissionPlanner.MAVLinkInterface - setMode Circle (7)
INFO MissionPlanner.MAVLinkInterface - doCommand cmd DO_SET_MODE 1 7 0 0 0 0 0
INFO MissionPlanner.MAVLinkInterface - set giveComport Base Thread current False new False
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
HUD 6 hz drawtime 0 gl True
bps 2222 loss 0 left 693 mem 143.9736328125 mav2 True sign False mav1 0 mav2 65 signed 0
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
HUD 7 hz drawtime 0 gl True
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
bps 2139 loss 0 left 693 mem 140.9384765625 mav2 True sign False mav1 0 mav2 61 signed 0
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
HUD 8 hz drawtime 0 gl True
bps 2139 loss 0 left 693 mem 143.7294921875 mav2 True sign False mav1 0 mav2 61 signed 0
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
bps 2139 loss 0 left 693 mem 140.3974609375 mav2 True sign False mav1 0 mav2 61 signed 0
HUD 6 hz drawtime 0 gl True
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
[... 151 lines cut ...]
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
bps 2151 loss 0 left 692 mem 139.8095703125 mav2 True sign False mav1 0 mav2 62 signed 0
HUD 6 hz drawtime 0 gl True
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
INFO MissionPlanner.MAVLinkInterface - doCommand cmd COMPONENT_ARM_DISARM 0 21196 0 0 0 0 0
INFO MissionPlanner.MAVLinkInterface - set giveComport Base Thread current False new True
INFO MissionPlanner.MAVLinkInterface - doCommand cmd resp COMPONENT_ARM_DISARM - ACCEPTED
INFO MissionPlanner.MAVLinkInterface - set giveComport  current True new False
INFO MissionPlanner.MAVLinkInterface - 1/4/2024 10:57:48 PM 6 Disarming motors
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 1105
route OnRender Large Graphics Path 415
bps 2215 loss 0 left 690 mem 142.9287109375 mav2 True sign False mav1 0 mav2 64 signed 0
HUD 6 hz drawtime 0 gl True
route OnRender Large Graphics Path 1105

IamPete1, 2024-01-07:
It seems like this has been resolved, https://discuss.ardupilot.org/t/pre-made-enviroment-for-building-ardupilot/82176/45

closed 2024-01-07 by IamPete1