ArduPilot/ardupilot #25923: ardupilot/modules/waf/waf-light - change the shebang to python3

kwikius, 2024-01-07:
## Bug report

**Issue details**

_Please describe the problem_

ardupilot/modules/waf/waf-light has as a shebang

#!/usr/bin/env python

building ardupilot may throws an error since ubuntu may assume this refers to python2.

I have to change the shebang to

#!/usr/bin/env python3

Then the build works fine. I assume that waf always uses python3 so being explicit about it should not cause an issue?

**Version**
_What version was the issue encountered with_
My ubuntu version
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy

**Platform**
[ *] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
all

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
all ( sitl, stm32 various)

**Logs**
_Please provide a link to any relevant logs that show the issue_

amilcarlucas, 2024-01-08:
It would be better to create a PullRequest instead of an issue

kwikius, 2024-01-08:
@amilcarlucas  Sure thing. I wanted to raise it first to see if there might be problems with it. As far as a pull request, waf being a submodule, I presumably need to submit the pull request to ardupilot/waf repository first?

kwikius, 2024-01-09:
fixed by https://github.com/ArduPilot/waf/pull/14

closed 2024-01-09 by kwikius