ArduPilot/ardupilot #25933: autotest TestLogDownloadMAVProxyNetwork flapping

davidbuzz, 2024-01-08:
## Bug report

**Issue details**

any idea why there's a flapping test on "sub " at the momnent..?

>>>> FAILED STEP: test.Sub at Mon Jan  8 05:55:10 2024
Failure Summary:
  test.Sub:
    TestLogDownloadMAVProxyNetwork (Download latest log over network port) (60) (see /tmp/buildlogs/ArduSub-TestLogDownloadMAVProxyNetwork.txt) (duration 108.54210472106934 s)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  X ] Submarine

**Logs**
```
...
Downloading log with UDPMulticast mcast:
...
2024-01-04T22:05:57.0872269Z Received 1392 parameters
2024-01-04T22:05:57.0872746Z Saved 1392 parameters to test.ArduSub/logs/2024-01-04/flight8/mav.parm
2024-01-04T22:05:57.0873230Z Timed out looking for Finished downloading
2024-01-04T22:05:57.0873617Z AT-0365.2: Exception caught: 120
2024-01-04T22:05:57.0873934Z Traceback (most recent call last):
2024-01-04T22:05:57.0874568Z   File "/w/ardupilot/ardupilot/Tools/autotest/vehicle_test_suite.py", line 8303, in run_one_test_attempt
2024-01-04T22:05:57.0875198Z     test_function(**test_kwargs)
2024-01-04T22:05:57.0875840Z   File "/w/ardupilot/ardupilot/Tools/autotest/vehicle_test_suite.py", line 4361, in TestLogDownloadMAVProxyNetwork
2024-01-04T22:05:57.0876583Z     mavproxy.expect("Finished downloading", timeout=120)
2024-01-04T22:05:57.0877241Z   File "/__w/ardupilot/ardupilot/Tools/autotest/pysim/util.py", line 704, in _expect_callback
2024-01-04T22:05:57.0877825Z     raise pexpect.TIMEOUT(timeout)
2024-01-04T22:05:57.0878176Z pexpect.exceptions.TIMEOUT: 120
```

... and mavproxy emits it here...
https://github.com/ArduPilot/MAVProxy/blob/2aeaaff6a6f40bfceb3c155c56c247b2dad09589/MAVProxy/modules/mavproxy_log.py#L70

actually... that action 'Timed out looking for numLogs'  with sub in the same basic area... 'TestLogDownloadMAVProxyNetwork'
2024-01-08T05:54:55.5393005Z AT-0330.3: Exception caught: 60
2024-01-08T05:54:55.5393594Z Traceback (most recent call last):
2024-01-08T05:54:55.5394771Z   File "/w/ardupilot/ardupilot/Tools/autotest/vehicle_test_suite.py", line 8303, in run_one_test_attempt
2024-01-08T05:54:55.5401203Z     test_function(**test_kwargs)
2024-01-08T05:54:55.5402456Z   File "/w/ardupilot/ardupilot/Tools/autotest/vehicle_test_suite.py", line 4359, in TestLogDownloadMAVProxyNetwork
2024-01-08T05:54:55.5403775Z     mavproxy.expect("numLogs")
2024-01-08T05:54:55.5404780Z   File "/__w/ardupilot/ardupilot/Tools/autotest/pysim/util.py", line 704, in _expect_callback
2024-01-08T05:54:55.5405846Z     raise pexpect.TIMEOUT(timeout)
2024-01-08T05:54:55.5406443Z pexpect.exceptions.TIMEOUT: 60
[07:28]
ok, so it looks like both of them are 'Exception caught: ' and 'pexpect.exceptions.TIMEOUT' in slightly different areas of sub's log download.

https://github.com/ArduPilot/ardupilot/actions/workflows/test_sitl_sub.yml?query=is%3Afailure
here's a few recent ones..
https://github.com/ArduPilot/ardupilot/actions/runs/7435994782/job/20232626254
https://github.com/ArduPilot/ardupilot/actions/runs/7443282050/job/20248632319
https://github.com/ArduPilot/ardupilot/actions/runs/7448925315/job/20276583962
https://github.com/ArduPilot/ardupilot/actions/runs/7435036262/job/20230061994
this particular one is in sub 'TCPServer' log download , so it appears to vary wether its UDP or TCP, and multicast or not....
https://github.com/ArduPilot/ardupilot/actions/runs/7426883926/job/20211966042
https://github.com/ArduPilot/ardupilot/actions/runs/7391265870/job/20110097203
https://github.com/ArduPilot/ardupilot/actions/runs/7384057875/job/20086347040
https://github.com/ArduPilot/ardupilot/actions/runs/7377533431/job/20077370944
the oldest I was able to find was.. on dec 30 ... but i didn't have a good look earlier, just spot chcks...
https://github.com/ArduPilot/ardupilot/actions/runs/7362324693/job/20040714628

tridge/peter tried to fix it by making hte logs smaller here:
```
commit 816e3fae3e90aba9773ededf2308bed5c1755eef
Author: Andrew Tridgell <andrew@tridgell.net>
Date:   Mon Jan 1 08:37:52 2024 +1100

    autotest: fixed flapping sub log download test
    
    the log being downloaded can be very large, and times out. Setting
    LOG_DISARMED=1 gives us a small log to download
```
here's a fresh, 'today' one on master: https://github.com/ArduPilot/ardupilot/actions/runs/7453063883/job/20277901701#step:7:18892

reviewing a "good" run , it shows that because of the above patch, logs are typically downloaded in under 10 seconds apart from multicast, which is under 30 secs...

```
Downloading log with UDPClient :16001
Finished downloading MAVProxy-downloaded-net-log-UDPClient.BIN (1703424 bytes 8 seconds, 198.8 kbyte/sec 0 retries)

Downloading log with UDPServer udpout:127.0.0.1:16002
Finished downloading MAVProxy-downloaded-net-log-UDPServer.BIN (1323008 bytes 6 seconds, 206.3 kbyte/sec 0 retries)

Downloading log with TCPClient tcpin:0.0.0.0:16003
Finished downloading MAVProxy-downloaded-net-log-TCPClient.BIN (196608 bytes 1 seconds, 180.1 kbyte/sec 0 retries)

Downloading log with TCPServer tcp:127.0.0.1:16004
Finished downloading MAVProxy-downloaded-net-log-TCPServer.BIN (196608 bytes 0 seconds, 225.1 kbyte/sec 0 retries)

Downloading log with UDPMulticast mcast:
Finished downloading MAVProxy-downloaded-net-log-UDPMulticast.BIN (2179072 bytes 23 seconds, 94.6 kbyte/sec 88 retries)

Downloading log with UDPBroadcast :16005
Finished downloading MAVProxy-downloaded-net-log-UDPBroadcast.BIN (1323008 bytes 6 seconds, 215.3 kbyte/sec 0 retries)
```

I think reducing the timeout to 30sec, and adding 2 or 3 retries to it might be the way to go... as when it works, it seems to work ok...?

davidbuzz, 2024-01-08:
'UDPMulticast.BIN (2179072 bytes 23 seconds, 94.6 kbyte/sec 88 retries)' - does mavproxy's 'UDPMulticast' implementation need work..?

peterbarker, 2024-07-31:
I believe this one is now stable.

I certainly haven't seen it fail recently.

OK to close this one, @davidbuzz  ?

peterbarker, 2024-11-24:
Closing this one - haven't seen it flash in a while, and I did fix some issues around this stuff.

closed 2024-11-24 by peterbarker