ArduPilot/ardupilot #25942: AP_Baro: guessed_ground_temperature updated on outside air temperature

MonkO, 2024-01-09:
If I understand correctly, the ground temperature should be extrapolated from the outside air temperature

https://github.com/ArduPilot/ardupilot/blob/eb10b8196466f4b53e8d018cb6c4cb37804d1992/libraries/AP_Baro/AP_Baro.cpp#L397C62-L397C62

IamPete1, 2024-01-09:
Is this a question? Or do you see a bug? Or a feature request?

MonkO, 2024-01-10:
I think here is bug
https://github.com/ArduPilot/ardupilot/blob/eb10b8196466f4b53e8d018cb6c4cb37804d1992/libraries/AP_Baro/AP_Baro.cpp#L397

get_external_temperature() - temperature on altitude (from airspeed sensor), but _guessed_ground_temperature must be temperature on ground.

Maybe it will be right
`_guessed_ground_temperature = get_external_temperature() + ISA_LAPSE_RATE * get_altitude();`

IamPete1, 2024-01-10:
Although its not obvious from that function, it is only possible to calibrate the baro if disarmed or the EKF thinks your on the ground. If we were going to allow calibration in the air, then your right we would have to try and extrapolate.

MonkO, 2024-01-10:
Now I understand. Thank you!

closed 2024-01-10 by MonkO