ArduPilot/ardupilot #25946: Pixhawk 6c mini external compass disappearing

ashwanisihag, 2024-01-10:
When we use pixhawk 6c mini with m9 or m10 the external compass disappeard after few flights. Its been reported in almost every system that external compass is missing. Internal is always detected but external disappears. May be because they are same Ic.

We have checked everything and all seems normal when we reinstall firmware. Its not a loose connection problem.

boba1234567890, 2024-01-28:
Hi I've also been having this problem but with my Pixhawk reinstalling the Sofware (ArduCopter v4.4.4) is not solving anything. 
even downgrading to (v4.4.3) it still doesn't work.

rmackay9, 2024-02-20:
@ashwanisihag, @boba1234567890,

Can you tell us exactly which compass module you are using?  Is it part of a GPS unit?  Perhaps provide a picture if possible.  Also if you have an onboard log that would be great.  You should be able to create a log even if the autopilot is not armed if you first set LOG_DISARMED = 1.

rmackay9, 2024-02-26:
@ashwanisihag @boba1234567890,

Sorry to pester but could you clarify which compass module is being used?

ashwanisihag, 2024-03-17:
Hi Randy 

Sorry I missed your message was bit busy with army..thanks for replying...Yes i will share photos of FC and GPS today. Its a big problem now as we have bought 1000 FC units and all have the same issue.

By the way its pixawk 6c mini and  M9 or M10 gps units from holybro.

Regards

ashwanisihag, 2024-03-17:
> @ashwanisihag, @boba1234567890,
>
> Can you tell us exactly which compass module you are using? Is it part of a GPS unit? Perhaps provide a picture if possible. Also if you have an onboard log that [...]

HI compass is  IST8310 in M9N or M10N

ashwanisihag, 2024-03-17:
Just tested

Same issue in 4.5 beta
[image]
[1 01-01-1970 05-30-00.zip](https://github.com/ArduPilot/ardupilot/files/14627139/1.01-01-1970.05-30-00.zip)

ashwanisihag, 2024-03-17:
Is this the issue? 

"The IST8310 is more complicated to auto detect because it has 4 different I2C slave addresses."
https://intofpv.com/t-discover-compass-ist8310-in-betaflight
Although its related to betyaflight code

rmackay9, 2024-03-17:
Hi @ashwanisihag,

Thanks for the feedback and links, I've re-opened the issue and hopefully we can reproduce it and/or just fix it.

tridge, 2024-03-19:
@ashwanisihag we do look for 4 different i2c addresses for probing:
https://github.com/ArduPilot/ardupilot/blob/Plane-4.5/libraries/AP_Compass/AP_Compass.cpp#L1232
if you have spare boards that reproduce this issue reliably you could send one to me
otherwise we may need Saleae pro traces of the i2c bus

tridge, 2024-03-19:
@ashwanisihag feel free to ping me on discord too

ashwanisihag, 2024-03-20:
> @ashwanisihag we do look for 4 different i2c addresses for probing: https://github.com/ArduPilot/ardupilot/blob/Plane-4.5/libraries/AP_Compass/AP_Compass.cpp#L1232 if you have spare boards that [...]

Sure I can send you one set , Please guide me how to send and at what address

ashwanisihag, 2024-03-20:
> @ashwanisihag feel free to ping me on discord too

Sure i will try

ashwanisihag, 2024-03-20:
> @ashwanisihag we do look for 4 different i2c addresses for probing: https://github.com/ArduPilot/ardupilot/blob/Plane-4.5/libraries/AP_Compass/AP_Compass.cpp#L1232 if you have spare boards that [...]

I have seen this code and tried to figure out but was not abe to figure it out.

rmackay9, 2024-03-26:
@ashwanisihag,

I'd like to get to the bottom of this if possible.  Could you ping Andrew Tridgell or me on Discord?  I think maybe your Discord id is different from your GitHub id?

rmackay9, 2024-04-01:
No updates for a week so I'll close this issue again.  We will see responses however.

ashwanisihag, 2024-04-05:
> No updates for a week so I'll close this issue again. We will see responses however.

I will create Discord id and will connect. Thanks

rmackay9, 2024-04-16:
I've received another ping that this issue is real.

Craig-1965, 2024-04-16:
With continued communications with Holybro I received the following Email 11 March 2024 (and nothing since then nothing)

**Hello,
 
We have found that the lost external compass on a Pixhawk 6c running Ardupilot connected with a M9N GPS is caused by the Ardupilot firmware.
We need some time to get it fixed.
Now please give it Px4 instead which is free of such an issue.
Will keep you updated.
 
Thanks!
 
 
 
Holybro**

vincentpoont2, 2024-04-16:
> With continued communications with Holybro I received the following Email 11 March 2024 (and nothing since then nothing)
>
> **Hello,
>
> We have found that the lost external compass on a Pixhawk 6c [...]

This issue indeed only occur with Ardupilot, but sorry that our rep tell you to "give it Px4 instead which is free of such an issue". 

We are working with @rmackay9 @tridge to fix this issue in Ardupilot. The hardware are already on the way to tridge, and we will provide update here when we have a solution. 

Vince
Holybro

Craig-1965, 2024-04-16:
Doesn't do anything for the units lost and other damages or the approach followed by the Support staff.

I'll keep the other units out of the air until addressed.

What doesn't make sense and as discussed with he support staff.  Without a UBEC powering the Servo rail it remails a lot more stable.  I have one unit that I personally fly and I have around 22 hours of flying on it without incident.  It just so happens that this Servo rail is powered by a 5A UBEC from Tarot and if I change the UBEC to a different brand the instability returns.  

Doesn't make sense I know but this is live experience (may or not be of value) and this unit is one of the best drones I have in use at the moment.

rmackay9, 2024-04-17:
@Craig-1965,

If you have a log of the compass failing/disappearing in flight we'd really like to see it.  If that can be done without flying it all the better, maybe set LOG_DISARMED = 1.

Craig-1965, 2024-04-17:
It will be **IMPOSSIBLE** to get the Logs for the ones where it happened in Flight..... 

Those 2 Drones, Cameras, Payloads etc are still GONE/LOST/MISSING.  They did a sudden direction change and both flew off at 120km/h+ Until the battery ran out and were never seen again.  The one went into Alt Hold about 3.5km from home and I managed to fly it back on the screen for about 100m and then it shot away at 90 degrees to flight path and I lost comms at 6.1km over a residential area.... Luckily only travelling at 18m/s when it went down.  I haven't seen reports of injury or property damage, but I am sure there was.

I'll just have to get the SD Card from the last used 6C Mini.  I will try the LOG_DISARMED = 1 and send the recorded log.  

I'm just out of country right now so will probably only get it done at the earliest next week.

tridge, 2024-04-30:
@ashwanisihag @Craig-1965 I have the hardware now, sent by @vincentpoont2 - what I need is a way to reproduce the issue. So far I've been testing on USB power, with numerous soft reboots to see if it ever comes up without the compass. I also tried having it sit for a long time to see if the compass would disappear. No luck so far reproducing the issue.
do either of you have suggestions on approached to triggering the issue?

Craig-1965, 2024-04-30:
It's a tricky one. I have 1 in a drone that I have now logged 400 flights with no issue and then the 2 that flew away were also working perfectly and then flew away.I have one but it's 1600km away. Will arrange for them to courier the unit to me with a shipment of drones that need to go to Australia after fine tuning.That unit is very badly behavedCraig Neil+27 82 448 7226

  disappearing  (Issue #25946)

tridge, 2024-04-30:
@Craig-1965 @ashwanisihag I have now managed to reproduce the issue and I've opened a PR that I think will fix it:
https://github.com/ArduPilot/ardupilot/pull/26929
I've put builds of plane and copter 4.5.1 with this fix here:
http://uav.tridgell.net/tmp/IST8310/
Please test!

Craig-1965, 2024-04-30:
Cannot access the .apj file

 

Tried on 3 PC in 2 different networks

rmackay9, 2024-04-30:
@Craig-1965,

I've checked [Tridge's link](http://uav.tridgell.net/tmp/IST8310/) and there does not appear to be any password required to download from the site.   Maybe you could try again with a different browser?

[image]

tridge, 2024-04-30:
@Craig-1965 the problem was likely your browser auto-converting to https and my servers SSL records being corrupt
please try this now:
https://uav.tridgell.net/tmp/IST8310/
I have also put them here:
https://drive.google.com/drive/folders/12VJm_4b4449iJcfqIM1okFV8p19D-_WU?usp=sharing

closed 2024-04-30 by tridge

Craig-1965, 2024-05-01:
Thank you. It might have been a network issue but I returned there shortly after and downloaded the file.Will this fix be included in all future firmware updates for the 6C Mini or should we stick to this firmware?Craig Neil+27 82 448 7226

  disappearing  (Issue #25946) 

https://uav.tridgell.net/tmp/IST8310/
https://drive.google.com/drive/folders/12VJm_4b4449iJcfqIM1okFV8p19D-_WU?usp=sharing

tridge, 2024-05-01:
@Craig-1965 if your testing goes well we will include it in the 4.5.3 release. We did not include in todays 4.5.2-beta as we don't yet have confirmation it fixes your issue. Also note that @vincentpoont2 from Holybro has indicated a hardware fix may be coming

ashwanisihag, 2024-12-15:
It seems like issue has been fixed in latest firmware's ( may be after 4.5.3 release). Am i right andrew?

rmackay9, 2024-12-15:
Hi @ashwanisihag,

Yes, [the fix from Tridge](https://github.com/ArduPilot/ardupilot/pull/26929) was included in the next release, 4.5.3

PR #26929 (2024-04-30): AP_Compass: workaround hardware bug in IST8310 whoami
the WAI (whoami) register is writeable. Not only is it writeable, but the written value is persistent across a power cycle. You have to remove power for about 30s for it to finally go back to the right default value of 0x10

this makes using WAI as a test for finding a IST8310 problematic. The best we can do is send a soft reset which will reset it to default for us to then check

fixes #25946