ArduPilot/ardupilot #25956: Protocol st24 work master but not stable

romucifer, 2024-01-11:
Hello rc protocol  st24 work in master firmware but not work in stable firmware 

 Input Rc protocol st24 in stable version

_Please describe the problem_
Rc protocol st24 not work in stable firmware 

**Platform**

[  ] Rover

**Airframe type**
Boat
Pixhawk 1m
**Logs**
_Please provide a link to any relevant logs that show the issue_

yuri-rage, 2024-01-11:
This is a user request to backport PR https://github.com/ArduPilot/ardupilot/pull/23742 to Rover stable (4.4).

See discussion: https://discuss.ardupilot.org/t/protocol-rc-st24-firmware-stable/111570

rmackay9, 2024-01-12:
OK, I've added it to the 4.4 project so it will be considered for 4.4.5.

closed 2024-01-12 by rmackay9

romucifer, 2024-01-12:
Thank youvery much

lipper-chengdu, 2024-01-15:
> OK, I've added it to the 4.4 project so it will be considered for 4.4.5.

https://github.com/ArduPilot/ardupilot/pull/23914
Can this PR be merged? Enable ST24 to support failsafe check and Status BackHaul RC.