ArduPilot/ardupilot #25958: Drone cannot be disarmed with HWSS (hardware safety switch) sometimes

ardugitter, 2024-01-11:
## Bug report

**Issue details**

**Setup:**
1.  Set DISARM_DELAY parameter to 0 so that once the drone is armed, the motors keep spinning until explicitly disarmed.
2. Configure motor emergency stop to be one of the spare channels on the transmitter.

**The issue:**
1. Arm drone by pressing HWSS on the drone.
2. Arm with joystick on transmitter. Motors will start spinning.
4. Set motor emergency stop using the transmitter. Motors will stop spinning.
5. Try to disarm by pressing the HWSS. It fails. It should succeed.

**Version**
Arducopter 4.3.8

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Normal X quadcopter

**Hardware type**
Pixhawk 1

**Logs**
Sorry don't have one at present.

rmackay9, 2024-01-12:
Hi, I suspect the issue is actually just that the vehicle does not think it is landed.  If you hold the throttle down and then try to disarm does it work?  I think it should.

BTW, it's best to report these in the forums first and then we can move them to the issues list once they're confirmed to be a real issue.  I'll leave this here for now but we need to be a little strict about this or our to-do list turns into another support forum.

ardugitter, 2024-01-12:
Yes, that works. Sorry, I'll post next time on the messaging board instead of assuming it's a bug.

It just seemed like a safety issue that using the HWSS should always allow a safe disarm, whatever the drone state. I'll close it for now and see what the messaging boards think. Thanks again.