ArduPilot/ardupilot #25962: the angular velocity of the body frame needs to be limited

lixiaowei123, 2024-01-12:
Issuse:
// Correct the thrust vector and smoothly add feedforward and yaw input
    _feedforward_scalar = 1.0f;
    if (_thrust_error_angle > AC_ATTITUDE_THRUST_ERROR_ANGLE * 2.0f) {
        _ang_vel_body.z = _ahrs.get_gyro().z;
    } else if (_thrust_error_angle > AC_ATTITUDE_THRUST_ERROR_ANGLE) {
        _feedforward_scalar = (1.0f - (_thrust_error_angle - AC_ATTITUDE_THRUST_ERROR_ANGLE) / AC_ATTITUDE_THRUST_ERROR_ANGLE);
        _ang_vel_body.x += ang_vel_body_feedforward.x * _feedforward_scalar;
        _ang_vel_body.y += ang_vel_body_feedforward.y * _feedforward_scalar;
        _ang_vel_body.z += ang_vel_body_feedforward.z;
        _ang_vel_body.z = _ahrs.get_gyro().z * (1.0 - _feedforward_scalar) + _ang_vel_body.z * _feedforward_scalar;
    } else {
        _ang_vel_body += ang_vel_body_feedforward;
    }
    //TODO:
    ang_vel_limit(_ang_vel_body, radians(_ang_vel_roll_max), radians(_ang_vel_pitch_max), radians(_ang_vel_yaw_max));
Why?
If the maximum angular velocity value is limited earlier, the feedforward angular velocity may exceed the maximum value limit.

Modify：

    ang_vel_limit(_ang_vel_body, radians(_ang_vel_roll_max), radians(_ang_vel_pitch_max), radians(_ang_vel_yaw_max));

amilcarlucas, 2024-01-12:
Pull requests are the preferred way to discuss code changes.

lixiaowei123, 2024-01-15:
[image]

void AC_AttitudeControl::attitude_controller_run_quat()
{
    // This represents a quaternion rotation in NED frame to the body
    Quaternion attitude_body;
    _ahrs.get_quat_body_to_ned(attitude_body);

    // This vector represents the angular error to rotate the thrust vector using x and y and heading using z
    Vector3f attitude_error;
    thrust_heading_rotation_angles(_attitude_target, attitude_body, attitude_error, _thrust_angle, _thrust_error_angle);

    // Compute the angular velocity corrections in the body frame from the attitude error
    _ang_vel_body = update_ang_vel_target_from_att_error(attitude_error);
    log_test_param1=_ang_vel_body.x*57.2957795;
    log_test_param2=_ang_vel_body.y*57.2957795;
    // ensure angular velocity does not go over configured limits
    ang_vel_limit(_ang_vel_body, radians(_ang_vel_roll_max), radians(_ang_vel_pitch_max), radians(_ang_vel_yaw_max));
    log_test_param3=_ang_vel_body.x*57.2957795;
    log_test_param4=_ang_vel_body.y*57.2957795;
    // rotation from the target frame to the body frame
    Quaternion rotation_target_to_body = attitude_body.inverse() * _attitude_target;

    // target angle velocity vector in the body frame
    Vector3f ang_vel_body_feedforward = rotation_target_to_body * _ang_vel_target;
    log_test_param5=ang_vel_body_feedforward.x*57.2957795;
    log_test_param6=ang_vel_body_feedforward.y*57.2957795;
    log_test_param7=ang_vel_body_feedforward.z*57.2957795;

    // Correct the thrust vector and smoothly add feedforward and yaw input
    _feedforward_scalar = 1.0f;
    if (_thrust_error_angle > AC_ATTITUDE_THRUST_ERROR_ANGLE * 2.0f) {
        _ang_vel_body.z = _ahrs.get_gyro().z;

    } else if (_thrust_error_angle > AC_ATTITUDE_THRUST_ERROR_ANGLE) {
        _feedforward_scalar = (1.0f - (_thrust_error_angle - AC_ATTITUDE_THRUST_ERROR_ANGLE) / AC_ATTITUDE_THRUST_ERROR_ANGLE);
        _ang_vel_body.x += ang_vel_body_feedforward.x * _feedforward_scalar;
        _ang_vel_body.y += ang_vel_body_feedforward.y * _feedforward_scalar;
        _ang_vel_body.z += ang_vel_body_feedforward.z;
        _ang_vel_body.z = _ahrs.get_gyro().z * (1.0 - _feedforward_scalar) + _ang_vel_body.z * _feedforward_scalar;

    } else {
        _ang_vel_body += ang_vel_body_feedforward;

    }
    log_test_param8=_ang_vel_body.x*57.2957795;
    log_test_param9=_ang_vel_body.y*57.2957795;
    log_test_param10=_ang_vel_body.z*57.2957795;

lixiaowei123, 2024-01-15:
[image]

void AC_AttitudeControl::attitude_controller_run_quat()
{
    // This represents a quaternion rotation in NED frame to the body
    Quaternion attitude_body;
    _ahrs.get_quat_body_to_ned(attitude_body);

    // This vector represents the angular error to rotate the thrust vector using x and y and heading using z
    Vector3f attitude_error;
    thrust_heading_rotation_angles(_attitude_target, attitude_body, attitude_error, _thrust_angle, _thrust_error_angle);

    // Compute the angular velocity corrections in the body frame from the attitude error
    _ang_vel_body = update_ang_vel_target_from_att_error(attitude_error);
    log_test_param1=_ang_vel_body.x*57.2957795;
    log_test_param2=_ang_vel_body.y*57.2957795;
    // ensure angular velocity does not go over configured limits
    ang_vel_limit(_ang_vel_body, radians(_ang_vel_roll_max), radians(_ang_vel_pitch_max), radians(_ang_vel_yaw_max));
    log_test_param3=_ang_vel_body.x*57.2957795;
    log_test_param4=_ang_vel_body.y*57.2957795;
    // rotation from the target frame to the body frame
    Quaternion rotation_target_to_body = attitude_body.inverse() * _attitude_target;

    // target angle velocity vector in the body frame
    Vector3f ang_vel_body_feedforward = rotation_target_to_body * _ang_vel_target;
    log_test_param5=ang_vel_body_feedforward.x*57.2957795;
    log_test_param6=ang_vel_body_feedforward.y*57.2957795;
    log_test_param7=ang_vel_body_feedforward.z*57.2957795;

    // Correct the thrust vector and smoothly add feedforward and yaw input
    _feedforward_scalar = 1.0f;
    if (_thrust_error_angle > AC_ATTITUDE_THRUST_ERROR_ANGLE * 2.0f) {
        _ang_vel_body.z = _ahrs.get_gyro().z;

    } else if (_thrust_error_angle > AC_ATTITUDE_THRUST_ERROR_ANGLE) {
        _feedforward_scalar = (1.0f - (_thrust_error_angle - AC_ATTITUDE_THRUST_ERROR_ANGLE) / AC_ATTITUDE_THRUST_ERROR_ANGLE);
        _ang_vel_body.x += ang_vel_body_feedforward.x * _feedforward_scalar;
        _ang_vel_body.y += ang_vel_body_feedforward.y * _feedforward_scalar;
        _ang_vel_body.z += ang_vel_body_feedforward.z;
        _ang_vel_body.z = _ahrs.get_gyro().z * (1.0 - _feedforward_scalar) + _ang_vel_body.z * _feedforward_scalar;

    } else {
        _ang_vel_body += ang_vel_body_feedforward;

    }
    log_test_param8=_ang_vel_body.x*57.2957795;
    log_test_param9=_ang_vel_body.y*57.2957795;
    log_test_param10=_ang_vel_body.z*57.2957795;
    if (_rate_bf_ff_enabled) {
        // rotate target and normalize
        Quaternion attitude_target_update;
        attitude_target_update.from_axis_angle(Vector3f{_ang_vel_target.x * _dt, _ang_vel_target.y * _dt, _ang_vel_target.z * _dt});
        _attitude_target = _attitude_target * attitude_target_update;
    }

    // ensure Quaternion stay normalised
    _attitude_target.normalize();

    // Record error to handle EKF resets
    _attitude_ang_error = attitude_body.inverse() * _attitude_target;
}

lixiaowei123, 2024-01-15:
![Uploading 1705311447274.png…]

lixiaowei123, 2024-01-15:
[image]

lthall, 2024-01-15:
Sorry but it isn't clear to me what you are trying to describe.

lthall, 2024-01-15:
I think your confusion is that we apply the limits to the attitude correction and the feed forward separately rather than the sum of the two. This is by design.

Given this issue doesn't contain any of the required information or describe an issue or feature request clearly, I am going to close this. If you can fix the first post to clearly state the problem and include the required information I am happy to open it again.

closed 2024-01-15 by lthall