ArduPilot/ardupilot #25974: RiscV64 compiler optimization warning in AP_MotorsUGV.cpp

tmagik, 2024-01-14:
## Bug report

**Issue details**

Compiling AP_MotorsUGV fails with the following error which appears to be a RiscV64 specific compiler optimization problem:

```
In member function ‘void AP_MotorsUGV::output_omni(bool, float, float, float)’,
    inlined from ‘void AP_MotorsUGV::output_omni(bool, float, float, float)’ at ../../libraries/AR_Motors/AP_MotorsUGV.cpp:909:6:
../../libraries/AR_Motors/AP_MotorsUGV.cpp:945:83: error: array subscript 8 is above array bounds of ‘float [4]’ [-Werror=array-bounds]
  945 |             output_throttle(SRV_Channels::get_motor_function(i), thr_str_ltr_out[i] * 100.0f * output_scale);
      |                                                                  ~~~~~~~~~~~~~~~~~^
compilation terminated due to -Wfatal-errors.
```

**Version**
Git master branch (commit d278ed3f2c1d9d1d3e578c8390024323bfa529da Jan 13)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Rover

**Hardware type**
N/A (in development)

**Logs**

The error can be easily reproduced on Ubuntu 22.04 with the following commands:

```
sudo apt install g++-12-riscv64-linux-gnu
git clone https://github.com/tmagik/ardupilot
git checkout 1f0ef5e40e7b20934c3fc28854794cd335e927cd
./waf configure linux --board=bbbminifire
git submodule update --init --recursive
./waf

```

magicrub, 2024-01-14:
[image]

I was going to test this but it seems my WSL Ubuntu can't figure out where to find gcc. Anyhow, I was going to test this simple change but maybe you can do it:

```
 - float thr_str_ltr_out[AP_MOTORS_NUM_MOTORS_MAX];
 + float thr_str_ltr_out[_motors_num];

```

tmagik, 2024-01-14:
That looks more correct, and appears to work. PR incoming.

Also look at this to probably figure out the configure failure:
ls -l /usr/bin/riscv64-linux-gnu-g++
lrwxrwxrwx 1 root root 24 Jan 13 14:17 /usr/bin/riscv64-linux-gnu-g++ -> riscv64-linux-gnu-g++-12

magicrub, 2024-01-14:
[image]

magicrub, 2024-01-14:
Looks like it needs `ls -l /usr/bin/riscv64-linux-gnu-g++-12`

rmackay9, 2024-01-16:
This isn't a bug though really.  There's no way that _motors_num can be greater than 4 and AP_MOTORS_NUM_MOTORS_MAX is defined as 4.

I'm not sure why the compiler is complaining but it's the compiler getting it wrong I think.

closed 2024-01-22 by rmackay9