ArduPilot/ardupilot #26002: AP_Logger: for loop for unit/multiplier name unsafe

shancock884, 2024-01-16:
## Bug report

**Issue details**

While getting my head round logger units, I spotted something in [libraries/AP_Logger/AP_Logger.cpp](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Logger/AP_Logger.cpp#L290).

The for loop in the AP_Logger::unit_name and AP_Logger::multiplier_name uses `i<unit_id` and `i<multiplier_id` respectively, but unit_id and multiplier_id are the character ID to search for (e.g. 'e' = 101), not the true array size.
Normally the for loop with break when a match is found anyway, but to be safe, I believe these should be replaced with `_num_units` and `_num_multipliers`.

e.g. code now:
```c++
/// return a unit name given its ID
const char* AP_Logger::unit_name(const uint8_t unit_id)
{
    for (uint8_t i=0; i<unit_id; i++) {
        if (_units[i].ID == unit_id) {
            return _units[i].unit;
        }
    }
    return nullptr;
}
``` 
Proposed change:
```c++
    for (uint8_t i=0; i<_num_units; i++) {
```

I don't mind creating a PR for this - its just that I am not sure how and when these functions get used in order to properly test it, beyond running a build to make sure it compiles.

**Version**
Ardupilot master branch

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
All (General logging)

**Hardware type**
All (General logging)

IamPete1, 2024-01-20:
Look like its a SITL only check, your fix looks correct to me, a PR would be great.

shancock884, 2024-01-22:
> Look like its a SITL only check, your fix looks correct to me, a PR would be great.

@IamPete1 - change as proposed is in PR #26059.

PR #26059 (2024-01-24): AP_Logger: Correct for loop condition
fix #26002
This PR corrects the for-loop condition expression in functions unit_name and multiplier_name of AP_Logger.cpp.
As described in the issue these were using the ID to search for (which could be something like: 'e' = 101), and not the actual size of the arrays, so could theoretically overrun the array when searching.
(Although as the loop with break when a match is found, it is pretty unlikely unless bad units/multipliers get used somehow).

SITL compiled and ran OK with the change.
closed 2024-01-24 by peterbarker