ArduPilot/ardupilot #26008: magVar Vector length threshold test

chjatala, 2024-01-17:
## Bug report
In the source code of [AP_RCTelemetry.cpp](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RCTelemetry/AP_RCTelemetry.cpp) there is a function **check_ekf_status**. In this function a Vector3f magVar is declared and the length of this vector is compared with a threshold to raise 'Error compass variance':
`if (magVar.length() >= 0.8f) {
        queue_message(MAV_SEVERITY_CRITICAL, "Error compass variance");
        check_ekf_status_timer = now;
}`
This comparison does not make sense.

IamPete1, 2024-01-17:
It is inline with what we do for the copter arming check.

https://github.com/ArduPilot/ardupilot/blob/15cea77e98d51a5371c38115ee56eb7a85ab26ff/ArduCopter/AP_Arming.cpp#L486-L489

What is the issue?

chjatala, 2024-01-17:
The issue is how the compass health can be checked by threshold comparison of the vector length.

chjatala, 2024-01-17:
Seeing that the vector3.length() function returns norm of the vector, the comparison makes sense.

closed 2024-01-17 by chjatala